roboticslab-uc3m/teo-main #2: Teo-Main PC out-of-date

jgvictores, 2017-03-09:
_From @jgvictores on March 7, 2017 18:6_

Hola,

Es posible que el Pc-teomain del laboratorio esté desactualizado y que algunas cosas no funcionen?? y en caso afirmativo, para cuando podría estar listo?? gracias

_Copied from original issue: roboticslab-uc3m/teo-software-manual#3_

jgvictores, 2017-03-09:
_From @rsantos88 on March 7, 2017 22:13_

Me imagino que te refieres a los nuevos repositorios que habeis cambiado. Mañana a primera hora me pongo con ello. Actualizo también los PCs de Teo.
Un saludo

jgvictores, 2017-03-09:
@rsantos88 Jejeje esto fue @jmgarciah con mi cuenta....

@jmgarciah Sí, hemos cambiado mucho software, hemos roto cosas pero todo es para mejor... Ahora hasta tenemos un repositorio de visión que se llama **vision**, uno de cinemática y dinámica llamado **kinematics-dynamics**... :smile:  Pronto estará todo funcional de nuevo... ¡estamos contigo!

jgvictores, 2017-03-09:
_From @PeterBowman on March 7, 2017 22:27_

Ya que ha surgido este tema, apunto también por aquí que quedan por arreglar algunas referencias en la wiki, que a su vez redundan en un cambio de configuración en los PCs que usáis y que por ese mismo motivo dejé como estaban: http://robots.uc3m.es/index.php/User:BartekPL/migraci%C3%B3n_repos. El comentario de `scripts/gnome/***.desktop` se refiere a que en [scripts/gnome/teo-main.desktop](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/845178b/scripts/gnome/teo-main.desktop) y similares hay rutas que no basta con cambiar en el código, también habrá que renombrar directorios allá donde este se use. Y por último, falta por redirigir las páginas de documentación generadas en robots.uc3m.es.

jgvictores, 2017-03-09:
@PeterBowman Síiii... aún nos queda mucho por hacer...

De hecho el nuevo https://github.com/roboticslab-uc3m/teo-main está totalmente en fase YCM-playground :sweat_smile:

rsantos88, 2017-03-09:
El PC teo-main se encuentra ya actualizado con el nuevo repositorio [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics), así como los PCs de teo-body con el repositorio [yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices) y teo-head con [vision](https://github.com/roboticslab-uc3m/vision). Los antiguos repositorios se han movido al directorio `moverADondeCorresponda`, pendientes de ser borrados cuando se considere oportuno. 

**Nota:** tanto en el PC de manipulación como en el de locomoción, estoy haciendo pruebas con otras versiones del repositorio, así que os aconsejo que si vais a trabajar con la nueva versión, la reinstaléis.

jgvictores, 2017-03-20:
El objetivo es tener cada repo descrito en [teo-software-manual -> overview -> PC software](https://github.com/roboticslab-uc3m/teo-software-manual/blob/ec73609cd630bcb53c8b1d0e5fd8039a73e3a8e4/overview-pc.md) hasta que no esté #3.

rsantos88, 2017-05-08:
Se puede cerrar esta issue??

jgvictores, 2017-05-08:
Para el PC `teo-main` hay dos alternativas (que no se excluyen mutuamente):
1. Simplemente este repo, que se baja todas las demás.
2. Cada repo del que depende, que está documentado por ejemplo en: https://github.com/roboticslab-uc3m/teo-main/blob/develop/doc/teo-main-install.md

rsantos88, 2017-11-27:
Acabo de actualizar el PC `teo-main` para que también tenga funcional la demo [teo-self-presentation](https://github.com/roboticslab-uc3m/teo-self-presentation).
Si existiese algún repositorio, aplicación o demo que falte por actualizar en este PC, comentarlo.
Cierro issue

closed 2017-11-27 by rsantos88

jgvictores, 2017-11-27:
@rsantos88 Grandeee!!

PS: Anyway, the issue description (made with my account but not by me) was way too broad. Thanks!!