roboticslab-uc3m/teo-main #3: Need functional version of this repo

jgvictores, 2017-03-09:
Need functional version of this repo.

jgvictores, 2017-03-20:
Assigning low priority, as high priority goes to having each individual repository work.

jgvictores, 2017-04-17:
Most repos build and have travis now. Currently breaking on the `openrave0.9-plugins` hack from ROBOTICSLAB_OPENRAVE_YARP_PLUGINS.

jgvictores, 2017-04-19:
Specifically:
```
-- Install configuration: ""
-- Installing: /home/yo/repos/teo-main/build/install/share/roboticslab-openrave-yarp-plugins/contexts/teoSim/teoSim.ini
-- Installing: /home/yo/repos/teo-main/build/install/bin/teoSim
-- Removed runtime path from "/home/yo/repos/teo-main/build/install/bin/teoSim"
-- Installing: /home/yo/repos/teo-main/build/install/lib/yarp/YarpOpenraveControlboard.so
-- Removed runtime path from "/home/yo/repos/teo-main/build/install/lib/yarp/YarpOpenraveControlboard.so"
-- Installing: /home/yo/repos/teo-main/build/install/share/yarp/plugins/YarpOpenraveControlboard.ini
-- Installing: /home/yo/repos/teo-main/build/install/lib/yarp/YarpOpenraveControlboardCollision.so
-- Removed runtime path from "/home/yo/repos/teo-main/build/install/lib/yarp/YarpOpenraveControlboardCollision.so"
-- Installing: /home/yo/repos/teo-main/build/install/share/yarp/plugins/YarpOpenraveControlboardCollision.ini
-- Installing: /home/yo/repos/teo-main/build/install/lib/yarp/FakeControlboard.so
-- Removed runtime path from "/home/yo/repos/teo-main/build/install/lib/yarp/FakeControlboard.so"
-- Installing: /home/yo/repos/teo-main/build/install/share/yarp/plugins/FakeControlboard.ini
-- Installing: /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so
CMake Error at openraveplugins/OpenraveYarpControlboard/cmake_install.cmake:50 (FILE):
  file INSTALL cannot copy file
  "/home/yo/repos/teo-main/build/external/ROBOTICSLAB_OPENRAVE_YARP_PLUGINS/lib/libOpenraveYarpControlboard.so"
  to "/usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so".
Call Stack (most recent call first):
  openraveplugins/cmake_install.cmake:37 (INCLUDE)
  cmake_install.cmake:40 (INCLUDE)

make[3]: *** [install] Error 1
make[2]: *** [external/ROBOTICSLAB_OPENRAVE_YARP_PLUGINS/CMakeFiles/YCMStamp/ROBOTICSLAB_OPENRAVE_YARP_PLUGINS-install] Error 2
make[1]: *** [CMakeFiles/ROBOTICSLAB_OPENRAVE_YARP_PLUGINS.dir/all] Error 2
```

PeterBowman, 2017-04-20:
Upstream issue regarding the OpenRAVE hack: roboticslab-uc3m/openrave-yarp-plugins#25.

PeterBowman, 2017-04-21:
> Upstream issue regarding the OpenRAVE hack: roboticslab-uc3m/openrave-yarp-plugins#25.

Fixed at roboticslab-uc3m/openrave-yarp-plugins@b26f7e9.

Now, the build process crashes on kinematics-dynamics (`error: ‘class KDL::ChainIdSolver_RNE’ has no member named ‘strError’`). A specific approach has been proposed at roboticslab-uc3m/kinematics-dynamics#40.

jgvictores, 2017-04-22:
I am closing this issue due to the fact that this repository can be compiled on a clean system.
1. The fact that YCM does not default to pass variables such as `cmake .. -Dorocos_kdl_DIR=/usr/local/share/orocos_kdl` to sub-repositories can make sense.
1. The fact that ROS `setup.bash` assigns environmental variables that affect compilation is not really an issue, as sourcing `setup.bash` can be temporarily omitted, or environmental variables reset, both solutions commented on roboticslab-uc3m/kinematics-dynamics#40.

Proceeding to work on #8: Add travis continuous integration!

closed 2017-04-22 by jgvictores