roboticslab-uc3m/teo-main #8: Add tests and continuous integration

jgvictores, 2017-04-06:
Add tests (e.g. via gtest) and continuous integration (e.g. via travis).

PeterBowman, 2017-04-22:
Once we have a functional Travis YAML file, I'd like to propose enabling [scheduled cron jobs](https://docs.travis-ci.com/user/cron-jobs/) to make this repo aware of changes in dependent (YCM-loaded) repositories.

jgvictores, 2017-04-22:
> Once we have a functional Travis YAML file, I'd like to propose enabling scheduled cron jobs to make this repo aware of changes in dependent (YCM-loaded) repositories.

I agree, that makes sense in this kind of super-build/meta-package repo. Good idea! :smile:

jgvictores, 2017-04-22:
Build passing at 03d4ac78c7b7a40ec36192bd4a3be7c897d3ab27.

jgvictores, 2017-04-22:
Have activated daily build on `master` via: https://travis-ci.org/roboticslab-uc3m/teo-main/settings

Closing issue!

closed 2017-04-22 by jgvictores