roboticslab-uc3m/teo-main #10: yarprun service on TEO network PCs does not always connect to teo-main yarp server

jgvictores, 2017-05-05:
_From @jgvictores on June 28, 2016 10:29_

The current use of daemontools makes yarprun run sometimes without connecting to the yarp server, which causes problems such as #42.
A `yarp detect` or similar (checking the return value) should be included in the daemontools service script (`/etc/service/yarprun`) or even completely replace daemontools for a different method to avoid this.

_Copied from original issue: roboticslab-uc3m/kinematics-dynamics#43_

jgvictores, 2017-05-05:
Should move this issue somewhere.

jgvictores, 2018-02-08:
Would affect services currently documented at https://github.com/roboticslab-uc3m/teo-developer-manual/blob/540b9521394a804a21e835bb0b0357fb369c8a67/appendix/b-teo-network-information.md

PeterBowman, 2021-07-14:
> (..) or even completely replace daemontools for a different method to avoid this.

I described a different solution at https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/59#issuecomment-753471971:

> On Ubuntu focal, I just added both `yarpserver` and `yarprun` entries to the GUI app `Startup Applications` and it worked ([ref](https://help.ubuntu.com/stable/ubuntu-help/startup-applications.html.en)). I believe this is nothing new in Ubuntu (already present in [Ubuntu 10.10](https://askubuntu.com/q/49274)), so why did we resort to using `daemontools` instead?

PeterBowman, 2021-11-09:
Today I'd rather use systemctl as in https://github.com/roboticslab-uc3m/yarp-devices/issues/251#issuecomment-919420954, though.