roboticslab-uc3m/teo-main #12: Try and doc teoBase with walking demo

jgvictores, 2017-05-06:
_From @jgvictores on October 14, 2016 8:55_

Try and doc teoBase with walking demo.
- [ ] Try
- [ ] Doc at http://robots.uc3m.es/index.php/Procedimiento_Demos

_Copied from original issue: roboticslab-uc3m/kinematics-dynamics#67_

jgvictores, 2017-05-06:
Requires https://github.com/roboticslab-uc3m/teo-main/issues/53 first.

jgvictores, 2017-05-06:
Originally set for NOV 01. :smile:

jgvictores, 2017-05-06:
_From @PeterBowman on May 3, 2017 16:52_

Blocked by #53.

PeterBowman, 2017-08-16:
Fixing broken ref in my previous comment: blocked by roboticslab-uc3m/kinematics-dynamics#53.

PeterBowman, 2021-11-09:
I believe teoBase.xml is not intended for specific demos, but rather for launching main controllers. Demo-related launchers should be hosted in their own XML. Anyway, there is no such thing as a "walking demo" as far as I know, besides the approach taken at https://github.com/roboticslab-uc3m/teo-main/issues/51.

closed 2022-06-04 by PeterBowman