roboticslab-uc3m/teo-main #15: Unify robot model joint limits

PeterBowman, 2017-08-16:
_From @jgvictores on April 21, 2016 17:17_

Should be in wiki, where work has been done (resulting in [current TEO Diagrams](http://robots.uc3m.es/index.php/TEO_Diagrams)). The following is a list of models to potentially be merged or linked somehow.
- [ ] Real joint limits are in [.ini of launch files in teo-body](https://github.com/roboticslab-uc3m/yarp-devices/tree/891b31117a7f703ff2f367832e42d36e8d164c86/share), should be the same across files.
- [ ] Kinematics used by [KdlSolver](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/8d798ca1b44df125fd404261ca04996641738a83/libraries/TeoYarp/KdlSolver): ([.ini files that contain DH parameters](https://github.com/roboticslab-uc3m/teo-configuration-files/tree/9f08cf2e0a121ab0d714bea6cdb8ed295de9593c/share/kinematics)). Perhaps could be removed from here?
- [x] Kinematics used by [teo_batc](https://github.com/roboticslab-uc3m/teo_batc): ([Gazebo model](https://github.com/roboticslab-uc3m/teo-gazebo)), ([ROS Gazebo model](https://github.com/roboticslab-uc3m/teo-ros-simulation)).
- [ ] Kinematics used by [teoSim](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/c4d034feed12e7ad010cebb38ca468e1d1054aca/programs/teoSim): ([Openrave model](https://github.com/roboticslab-uc3m/teo-openrave-models/tree/65b36e548bc25e7d4c413c822df9819c414bbfef/openrave/teo)).

*Updated on 16/08/17 by PeterBowman.*

_Copied from original issue: roboticslab-uc3m/kinematics-dynamics#31_

PeterBowman, 2017-08-16:
_From @RaulFdzbis on May 3, 2016 12:34_

[Gazebo model](https://github.com/roboticslab-uc3m/teo-gazebo) and [Ros Gazebo model](https://github.com/roboticslab-uc3m/teo-ros-simulation), were already updated in [7718f22](https://github.com/roboticslab-uc3m/teo-gazebo/commit/7718f22c90b09daludo.c30c5b4b4b55e1aa86e3c85566) and [a6199bc](https://github.com/roboticslab-uc3m/teo-ros-simulation/commit/a6199bc965c44919686355d27d876d9e002fa77c) with the joint limits used by the KdlSolver in [teo_main](https://github.com/roboticslab-uc3m/teo-main/tree/master/share/kinematics).

jgvictores, 2017-12-07:
Blocked by https://github.com/roboticslab-uc3m/teo-software-manual/issues/10 as the entry point will be teo-software-manual

jgvictores, 2017-12-13:
Unblocked. We should put the official values at: https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/appendix/a-teo-diagrams.md

jgvictores, 2017-12-13:
Have to consult if the joint limits currently at: https://github.com/roboticslab-uc3m/teo-configuration-files/tree/develop/share/kinematics are the ones that should be copied to the official value place: https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/appendix/a-teo-diagrams.md

rsantos88, 2017-12-18:
I saw that those limits (at https://github.com/roboticslab-uc3m/teo-configuration-files/tree/develop/share/kinematics) are not the same as the `.ini` values

jgvictores, 2017-12-18:
Not very sure to what you refer. Updated top description:

Robot model joint limits reference should be in https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/appendix/a-teo-diagrams.md

The following is a list of models to potentially be updated.
- `.ini` of launch files: https://github.com/roboticslab-uc3m/teo-configuration-files/tree/develop/share/launchManipulation and https://github.com/roboticslab-uc3m/teo-configuration-files/tree/develop/share/launchLocomotion
- `.ini` of kinematics files: https://github.com/roboticslab-uc3m/teo-configuration-files/tree/develop/share/kinematics
- Openrave models: https://github.com/roboticslab-uc3m/teo-openrave-models
- Gazabo models: https://github.com/roboticslab-uc3m/teo-gazebo-models, [teo_batc](https://github.com/roboticslab-uc3m/teo_batc): ([Gazebo model](https://github.com/roboticslab-uc3m/teo-gazebo)), ([ROS Gazebo model](https://github.com/roboticslab-uc3m/teo-ros-simulation)).
- ArmarX models: https://github.com/roboticslab-uc3m/teo-simox-models

jgvictores, 2017-12-18:
Note that maximum velocities and accelerations are also relevant!

jgvictores, 2017-12-18:
Blocked, because ideally we would get these values from the real robot, which is currently being modified, see issues on https://github.com/roboticslab-uc3m/teo-hardware-manual

AlvaroMartinezR, 2017-12-20:
What a coincidence, I was talking about the limits of TEO yesterday with @smcdiaz and he said thet the best I idea will be to test and obtain the values from the actual real robot like @jgvictores said:

>Blocked, because ideally we would get these values from the real robot, which is currently being modified, see issues on https://github.com/roboticslab-uc3m/teo-hardware-manual

The actual status of the robot only allow us to operate with locomotion, he has no arms. @rsantos88 and me are going to do it. 

>ArmarX models: https://github.com/roboticslab-uc3m/teo-simox-models

PS: I am responsible for `teo-simox-models` so next time assign or mention me please.

jgvictores, 2017-12-20:
> PS: I am responsible for `teo-simox-models` so next time assign or mention me please.

@AlvaroMartinezR No problem! Just in case, I would also recommend you to [watch](https://github.com/roboticslab-uc3m/teo-main/watchers) this repo. :-)

AlvaroMartinezR, 2018-01-19:
Blocked (again) by https://github.com/roboticslab-uc3m/teo-main/issues/29

rsantos88, 2018-01-24:
@AlvaroMartinezR and me have collected all the mechanical limits from the real robot.
See the table below:
## Joint mechanical limits

| Limb            | Link                                               | Upper limit | Lower limit |
|:----------------|:---------------------------------------------------|:--------|:-----|
| Right arm| 1                                                  | 103.075569 | -93.479797    | 
|                 | 2                                                  | -80.474518 | 34.622143    | 
|                 | 3                                                  | 85.149384 | -62.021088 | 
|                 | 4                                                  | 88.488579 | -37.943756  | 
|                 | 5                                                  | 79.701233 | -54.376099  | 
|                 | 6                                                  | 112.478027 | -61.933228   | 
| Left arm  | 1                                                  | -98.488586 | 92.618629  | 
|                 | 2                                                  |79.525482 | -34.253082 | 
|                 | 3                                                  | -87.504395 | 59.753952   | 
|                 | 4                                                  | -106.133575 |19.156414 | 
|                 | 5                                                  | -81.441132 | 74.604568  | 
|                 | 6                                                  | -108.945526 | 64.850616 | 

Also with the plate for teo-waiter:

| Limb            | Link                                               | Upper limit | Lower limit |
|:----------------|:---------------------------------------------------|:--------|:-----|
| Left arm  | 6 with plate                                       | -57.803162 | 65.641472  | 

Now, we are going to polish them: lower them and make them symmetric. Lower them because these are the absolute mechanical limits, working with them will be quite dangerous for TEO.
You can see the actual results [here](https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/appendix/a-teo-diagrams.md#joint-mechanical-limits).
The joint limits are quite high. What do you think about them? 
For example:

| Limb            | Link                                               | Upper limit | Lower limit |
|:----------------|:---------------------------------------------------|:--------|:-----|
| Right arm| 1 (Raw values)                                                 | 103.075569 | -93.479797    | 
| Right arm       | 1 (Edited)                                                 | 90 | -90    |

jgvictores, 2018-01-28:
Proposal: hard limits vs soft limits, documenting both.

AlvaroMartinezR, 2018-01-29:
@jgvictores Yeah, that was the idea, but my question was if you think the thresholds are acceptable.

jgvictores, 2018-01-29:
I'd ask @smcdiaz 's opinion on that.

jgvictores, 2018-01-29:
Spoken with @smcdiaz Questionfir starters: Were these values taken via the absolute or relative encoders?

jgvictores, 2018-01-29:
A more detailed explanation of what I spoke with @smcdiaz : 
1. Determine number of relevant decimal values (depends on abs/rel encoder).
2. Establish a fixed tolerance values.

Example: `103.075569 | -93.479797` -> 1 decimal -> `103.1 | -93.5` -> 5 degrees -> `98.1 | -88.5`.

jgvictores, 2018-01-30:
((WIP, but legs blocked by issues such as https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/1 from private repo))

jgvictores, 2018-01-30:
PD: Can still work on [this](https://github.com/roboticslab-uc3m/teo-main/issues/15#issuecomment-361317177).

rsantos88, 2018-01-30:
> Spoken with @smcdiaz Question for starters: Were these values taken via the absolute or relative encoders?

This values have been taken with absolute encoders using torque mode

jgvictores, 2018-01-30:
Okay, so this means we shouldn't/can't take the transmission reduction into account.

From [the current BOM](https://github.com/roboticslab-uc3m/teo-hardware-manual/blob/f9c7305b886cd8a43111a11bff57f2023ca14641/bill-of-materials.md) we have raw CUI AMT 203-V on that side. How much was that, (1024 * 4) pulses/turn? [/me a bit in a hurry to check yarp-devices source code]

rsantos88, 2018-01-31:
> From the current BOM we have raw CUI AMT 203-V on that side. How much was that, (1024 * 4) pulses/turn?

Yes, from https://github.com/roboticslab-uc3m/teo-configuration-files/blob/762ebe5079e05da38602e21e2feccd9901d8513d/share/launchManipulation/conf/launchManipulation.ini we have 4096 (1024 * 4)

rsantos88, 2018-01-31:
Which means 0.088 degree/pulse, so it's safe to say we keep only 1 decimal.

rsantos88, 2018-02-01:
I am working in this table with the definitive values [here](https://github.com/roboticslab-uc3m/teo-hardware-manual/blob/unify-joints/assets/motores-motores.csv).
We should remove [this](https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/appendix/a-teo-diagrams.md#joint-mechanical-limits) other to avoid duplicate information.

jgvictores, 2018-02-02:
Small notice: As a result of https://github.com/roboticslab-uc3m/teo-developer-manual/issues/27 we have merged teo-software-manual and teo-hardware manual, so progress will continue at https://github.com/roboticslab-uc3m/teo-developer-manual

Related: https://github.com/roboticslab-uc3m/teo-developer-manual/issues/33

rsantos88, 2018-02-05:
updated [.osd](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/issue%2315-joint-limits/assets/src/motores.ods) file with limit values and exported [here](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/issue%2315-joint-limits/assets/motores-motores.csv) in CSV.
The values with (*) has been replicated from the other leg

AlvaroMartinezR, 2018-02-06:
I have a proposal about the hard/soft limits found [here](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/issue%2315-joint-limits/assets/motores-motores.csv). 
-First, compare and unify opposite joints, for example: left and right saggital shoulder should have the same limit.
-Second, I personally find the limits, even they have been measured and can be phisically achieved are too wide for working with them.

These are my suggestions and ideas, I would like to know what do you guys think.

jgvictores, 2018-02-06:
There is a school of thought that believes robots should not be limited to what humans can do, which applies even to humanoid robotics. Imposing human restrictions for reasons other than pure safety would not respect this belief.

I'd agree if these extra artificial were placed in separate new column(s), which could be called as in "human-inspired joint limits" or similar, but not replacing the existing hard/soft limit columns. Regarding anybody declaring such new columns as useless/redundant/non-informative, perhaps they could be justified if they clearly referenced some kind of medical citation.

AlvaroMartinezR, 2018-02-12:
Totally agree, I actually found [this,](https://es.slideshare.net/pedroespinosa106/52756563-angulosyarcosdemovimiento) in order to reason the human-inspired limits with medical evidence.

rsantos88, 2018-02-14:
updated table [here](https://github.com/roboticslab-uc3m/teo-developer-manual/commit/d14a2b8c1ca855b6b048e03feac11df4fe8ed91e) with new "human-inspired joint limits" 
See the result in `.csv` format [here](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/issue%2315-joint-limits/assets/motores-motores.csv)

jgvictores, 2018-02-14:
Looks good! PR? :-)

jgvictores, 2018-02-14:
Well, beware: https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/10#issuecomment-365375610

jgvictores, 2018-02-15:
Merged https://github.com/roboticslab-uc3m/teo-developer-manual/commit/b95ffbf93f2d83c28ebe93eeba34820d7c6d3df8 so branches wouldn't get too messy, but still lots to do here.

Reminder to self: we are working at https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/assets/src/motores.ods and then updating https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/assets/motores-motores.csv

jgvictores, 2018-02-15:
Point to joint limits at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/7613ae9db87ffc548c93ead66e570659789462e4

rsantos88, 2018-02-16:
**note:** the legs values with asterisk has been replicated from the joints of the other leg

AlvaroMartinezR, 2018-02-16:
Ok, now that we have all the values, which ones are we going to use? I mean, this issue is called "unify joint limits", therefore, now that we have all the data, which ones do we put in the inis?

jgvictores, 2018-02-16:
I'd recommend putting the `Soft. Limits` in all the files listed [above](#issue-250572058). :-)

rsantos88, 2018-02-19:
> - [ ] Openrave models: https://github.com/roboticslab-uc3m/teo-openrave-models

Done at https://github.com/roboticslab-uc3m/teo-openrave-models/pull/17

rsantos88, 2018-02-19:
> - [x] `.ini` of launch files: https://github.com/roboticslab-uc3m/teo-configuration-files/tree/develop/share/launchManipulation and https://github.com/roboticslab-uc3m/teo-configuration-files/tree/develop/share/launchLocomotion

Done at https://github.com/roboticslab-uc3m/teo-configuration-files/pull/5

rsantos88, 2018-02-19:
@jgvictores pending of clean some unused files in https://github.com/roboticslab-uc3m/teo-configuration-files/tree/develop/share/kinematics

AlvaroMartinezR, 2018-02-20:
@PeterBowman Gazebo models seem quite forgotten, so ask directly to @RaulFdzbis or forget about it

PeterBowman, 2018-02-20:
Not my business, I just moved this issue :). Ping OP (@jgvictores).

AlvaroMartinezR, 2018-02-20:
Haha ok my mistake then, anyways I didnt meant it to sound mean, only to note that I found it useless if they are not used. Sorry

jgvictores, 2018-02-21:
Spoke with @RaulFdzbis today. He's in!

jgvictores, 2018-02-21:
@RaulFdzbis Please link to this issue after updating the Gazebo model!

@PeterBowman I'd like to discuss the possibility of removing joint limits from the kinematics `.ini` used from the solver:
- Reason of being is historic (iCub iKin kinematics file compatibility).
- Reason to remove is because it is duplicate info -> harder to maintain (specially with [several versions of kinematics](https://github.com/roboticslab-uc3m/teo-configuration-files/tree/ee168eaf61454113b1ac7113fbb24e10af679bc3/share/kinematics), potentially even after future cleaning).
- Implications are that users like @jmgarciah at teo-waiter apps have to remember to ask the robot for the joints and configure the solver if it is considered necessary (defaults are +-180 deg). I think this is not a bad thing, but maybe some warnings should be set. What is your opinion?

RaulFdzbis, 2018-02-21:
So I will update both [Gazebo](https://github.com/roboticslab-uc3m/teo-gazebo/tree/master/teo_model) and [ROS](https://github.com/roboticslab-uc3m/teo-ros-simulation/tree/master/teo_description) to the new joint values ( I suppose they have changed since [here](https://github.com/roboticslab-uc3m/teo-main/issues/15#issuecomment-322722674).

The Gazebo model will not be a problem, however modifying the URDF files for ROS could take a little bit of time so be patient! :D

PeterBowman, 2018-02-21:
> @PeterBowman I'd like to discuss the possibility of removing joint limits from the kinematics .ini used from the solver:

Those .ini values are not used at all, so IMO it's totally reasonable to remove them. Speaking of current cartesian controllers, I'd consider opening the robot device before the solver one, pick actual robot joint limits via `IControlLimits` interface, and pass them later to the solver upon configuration (e.g. `solverOptions.put("min", "whatever")`).

One thing to sort out is: should joint limits be constrained as read-only properties in `ICartesianSolver`?

* Yes, we set them upon device initialization and never modify their values again. `ICartesianSolver::setLimits` could be either removed or rewritten to hold the corresponding class invariants (= report failure if called more than once, which is perhaps a bit against the purpose of said method).
* No, keep `ICartesianSolver::setLimits` so that users may call it at will on runtime, but take some care to ensure it'll never fall into race conditions. Similarly, variables holding the kinematic chain have been guarded against such circumstances, recently.

jgvictores, 2018-02-21:
@PeterBowman Thanks for confirming that these values are not currently used by the solver(s)! I'll follow the conversation you started at https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/145#issuecomment-367329490.

Ok, to recap on the initial issue:
- [ ] `.ini` of launch files: @jgvictores must review https://github.com/roboticslab-uc3m/teo-configuration-files/pull/5
- [ ]`.ini` of kinematics files: @rsantos88 can safely delete the joint limit data.
- [ ] Openrave models: @jgvictores must review https://github.com/roboticslab-uc3m/teo-openrave-models/pull/17
- [ ] Gazabo models: will be done by @RaulFdzbis at as a low priority task, tracked at https://github.com/roboticslab-uc3m/teo-gazebo/issues/5
- [ ] ROS URDF model: will be done by @RaulFdzbis at as a low priority task, tracked at https://github.com/roboticslab-uc3m/teo-ros-simulation/issues/4
- [ ] ArmarX models: @AlvaroMartinezR maybe you could take a look?

AlvaroMartinezR, 2018-02-23:
@jgvictores Yes, but I am currently working on a paper for iros and since no one uses simox models I will update it as soon as I finish the article.

rsantos88, 2018-02-23:
> - [ ] `.ini` of kinematics files: @rsantos88 can safely delete the joint limit data.

I've removed the limits of [this](https://github.com/roboticslab-uc3m/teo-configuration-files/pull/6) files. Are the other files useful? Do you want to delete them? (e.g: `leftArm22Kinematics.ini` or `leftArmKinematics-isolated.ini`)

jgvictores, 2018-02-23:
> I've removed the limits of this files

@rsantos88 Thanks! I've merged https://github.com/roboticslab-uc3m/teo-configuration-files/pull/6

> Are the other files useful? Do you want to delete them? (e.g: leftArm22Kinematics.ini or leftArmKinematics-isolated.ini)

@rsantos88 Let's leave them for now. Stared https://github.com/roboticslab-uc3m/teo-configuration-files/issues/7 to track that issue.

> @jgvictores Yes, but I am currently working on a paper for iros and since no one uses simox models I will update it as soon as I finish the article.

@AlvaroMartinezR Thanks a lot, no hurry!

---

@rsantos88 Before merging the other PRs, I'm worried about recent changes and also the values marked with an asterisk. Postponing merges till we can speak about it in person!

AlvaroMartinezR, 2018-03-06:
- [ ]  TEO ArmarX Models  https://github.com/roboticslab-uc3m/teo-simox-models
I want to note that it would be better if we keep the symmetry for the soft and human inspired limits. Not pairing opposite joints does not make sense to me.

rsantos88, 2018-05-21:
> An error has been found in the maximum and minimum limits of the right and left axial wrists.
Changes updated in csv and ods file.
This change has to be taken into account for the rest of the .ini files

Fixed this wrong values at:
- [x] teo-openrave-models: https://github.com/roboticslab-uc3m/teo-openrave-models/commit/22c963c6cda0dac8438fca387245c51a8bd3b7ba
- [x] launchManipulation.ini: https://github.com/roboticslab-uc3m/teo-configuration-files/commit/d705af43ddaa899207bc1f2eb345cabe7ee7347f

@AlvaroMartinezR , remember to change this values in your ArmarX models

AlvaroMartinezR, 2018-05-22:
@rsantos88 Done

PeterBowman, 2018-10-16:
We are adding a small tolerance to joint limits in the cartesian controller, @jgvictores suggests that it could be taken into account here (https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/161#issuecomment-430241518).

jgvictores, 2018-12-12:
Blocked by #39

rsantos88, 2019-02-04:
After checking that the limits of the right frontal elbow was poorly annotated, we proceeded (thanks to @jmgarciah) to review all the joint limits of the arms, one by one, and made the respective modifications of some.
Changes updated in [csv](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/limits-review/assets/motores-transmision.csv) and [ods](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/limits-review/assets/editable/motores.ods) file. ([commit](https://github.com/roboticslab-uc3m/teo-developer-manual/commit/77694c4c93389c9bb01fd200b2902a42c8d64f77) here)
If you are agree, we can we can modify the .ini and the openrave files.
Sorry for the mess.

jgvictores, 2019-02-06:
No problem! As commented :smiley:2:smiley:, make all changes required (e.g. touch simulator if needed, not as part of a final solution, because still blocked by #39). :+1:

rsantos88, 2019-02-06:
Perfect!! Just to keep in mind that I'm using currently those limits in TEO, installed in the manipulation computer, making use of the branch [new-limits-reviewed](https://github.com/roboticslab-uc3m/teo-configuration-files/tree/new-limits-reviewed) :+1:

jgvictores, 2019-02-25:
This issue will be tedious, but a also a great opportunity to establish several ranges of joint limits (physical, driver-level recommended, software-level recommended, human-like...)!

[12 more replies]

closed 2019-07-04 by PeterBowman