roboticslab-uc3m/teo-main #16: Unify robot model dynamics

PeterBowman, 2017-08-16:
_From @jgvictores on April 20, 2016 10:59_

Issue inherited from wiki, where work has been done (resulting in [current TEO Diagrams](http://robots.uc3m.es/index.php/TEO_Diagrams)). The following is a list of models to potentially be merged or linked somehow.
- [ ] Kinematics(DH)+Mass/Inertia used by [KdlSolver](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/8d798ca1b44df125fd404261ca04996641738a83/libraries/TeoYarp/KdlSolver): [[1]](https://github.com/roboticslab-uc3m/teo-configuration-files/tree/9f08cf2e0a121ab0d714bea6cdb8ed295de9593c/share/kinematics).
- [ ] OpenRAVE 3D and Kinematics used by [teoSim](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/c4d034feed12e7ad010cebb38ca468e1d1054aca/programs/teoSim): [[2]](https://github.com/roboticslab-uc3m/teo-openrave-models/tree/b165a392a8700b73c9baf5bff7ea3caf114564b8/openrave/teo)
- [ ] Gazebo 3D and Kinematics+Mass/Inertia used by [teo-ros-simulation](https://github.com/roboticslab-uc3m/teo-ros-simulation): [[3]](https://github.com/roboticslab-uc3m/teo-ros-simulation/tree/fe3147dc5c5638d8239b7c8417a58b2eb5ac9185/teo_description)

For future reference, links to Mass/Inertia (official based on mechanical schematics): [.XLS](http://robots.uc3m.es/svn/TEOrepo/modelling/dynamics/Inercias_090518.xls) and [.PDF](http://robots.uc3m.es/svn/TEOrepo/modelling/dynamics/Inercias_090518.pdf).

*Updated on 16/08/17 by PeterBowman.*

_Copied from original issue: roboticslab-uc3m/kinematics-dynamics#29_

jgvictores, 2017-11-08:
> From @AlvaroMartinezR on November 8, 2017 12:15 at #21 :
> I have been working with SolidWorks and other CAE programs in order to develop an accurate dynamic model for TEO, including masses, inertias, CoM. It is not yet finished, but I wonder in which repository this kind of information (probably a .doc or a .pdf file) could be placed into.

There are a number of options on **where** to centralize this information (namely: teo-configuration-files, teo-hardware-manual, teo-software-manual).

This is argueable, but I currently recommend to create a markdown page (generated via [ods2md](https://github.com/kennytm/ods2md)) similar to [this one that got lost in translation](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/90497d45a6991862a59b56579f04d893fb65b566/doc/fig/DH%20parameters.md) as Appendix B of [https://github.com/roboticslab-uc3m/teo-software-manual](https://github.com/roboticslab-uc3m/teo-software-manual) (then upload the `.ods` to a `src` folder or similar).

The workflow then would be to update through other repositories (including teo-configuration-files, as well teo-openrave-models etc) that use their own program/library specific formats.

jgvictores, 2017-11-08:
PS: Assigned to @AlvaroMartinezR but there will be a number of future collaborators to propagate the updates through the corresponding repositories. Thanks!

AlvaroMartinezR, 2017-11-09:
Thanks! I will first finish extracting the information and then create a proper document.

jgvictores, 2017-11-09:
Great, thanks!

jgvictores, 2017-11-10:
I've been cleaning https://github.com/roboticslab-uc3m/teo-software-manual so now the workflow and placements are documented at: https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/CONTRIBUTING.md#workflow-for-assets

Hope that helps!

jgvictores, 2017-11-15:
Please note that Workflow and Placements (same rules) are now correctly at https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/README.md#workflow-for-assets

AlvaroMartinezR, 2017-11-20:
I finally obtained all dynamic information about all TEO's parts. SolidWorks provides the information in the following form: (see image below), as a plain text.
[image] (This comes straight out from SolidWorks, some lines are not usable in TEO's context, so will be edited or deleted)

My main inconvenient is that this kind of information I think it wouldn't fit well on a table similar to point 3 of the [workflow](https://github.com/roboticslab-uc3m/teo-software-manual/blob/master/README.md#workflow-for-assets). Therefore I think it will be better in a PDF.

AlvaroMartinezR, 2017-11-20:
After talking with Santiago, he told me to upload the info as a PDf in the doc folder of teo-simox-models and link it from the wiki.

jgvictores, 2017-11-20:
The screenshot is not plain text, it's `.odt`. What is the real output format? Is that copied and pasted?

AlvaroMartinezR, 2017-11-21:
Yes, it copies to your clipboard as plain text, I pasted it on a .ods file because you go piece by piece, so I needed to store the previous ones somewhere.

jgvictores, 2017-11-21:
Please add your `.odt` to the `assets/src` folder of https://github.com/roboticslab-uc3m/teo-software-manual. It would be great if you could add some lines at the top describing that it came from SolidWorks and link to the model.

After you do that, notify me here and I will proceed to see if `pandoc` does a good job in exporting `.ods` to `.md`. `.md` has very good properties (better parsed for search, better copied and pasted). Please note that an integrated `.pdf` is ultimately generated via the enabled GitBook mechanism (see [here](https://www.gitbook.com/book/roboticslab-uc3m/teo-software-manual/details), the Download PDF button).

jgvictores, 2017-11-21:
PS just because you mentioned `.ods` even though I got `.odt` from your screenshot:
- `.odt`: Equivalent to Word Document. We'd try `pandoc`.
- `.ods`: Equivalent to Excel Spreadsheet. This tabled format would be even better, and we could use `ods2md`.

jgvictores, 2017-11-21:
PPS: @AlvaroMartinezR Think you could give it a shot at pasting the parts into an `.ods`? Perhaps it could work out.

AlvaroMartinezR, 2017-11-21:
Well, dyslexia coming in place: I was all the time referring to `.odt`. In the meantime I was thinking in using md directly, actually I already started. So if I do it directly I save you this work. I will go for `.md` then.

jgvictores, 2017-11-21:
> So if I do it directly I save you this work. I will go for `.md` then.

`.md` is the new black. Three cheers for `.md`. What else can I say... Thanks! :+1:

jgvictores, 2017-11-21:
PD: Would go directly in `assets` for now. Still looking for a good way to `#include` one `.md` in another.

AlvaroMartinezR, 2017-11-22:
Done, created in `assets`.

jgvictores, 2017-11-22:
Great! Did some renaming at https://github.com/roboticslab-uc3m/teo-software-manual/commit/b95b219d51ae36eddb0839ccb9653b76752d810b

Still plenty to do to close this issue, just to recap from the top:
> - Kinematics(DH)+Mass/Inertia used by [KdlSolver](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/8d798ca1b44df125fd404261ca04996641738a83/libraries/TeoYarp/KdlSolver): [...]

jgvictores, 2017-11-22:
Also pending to resolve duplicates between https://github.com/roboticslab-uc3m/teo-simox-models and https://github.com/roboticslab-uc3m/teo-software-manual

AlvaroMartinezR, 2018-02-13:
@jgvictores Duplicates solved

jgvictores, 2018-02-13:
Still seeing some duplicates:
- https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/assets/teo-simox-names.pdf ([permalink](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/f3babf45f65ab6762314ca09ce2fffb190a4ba9b/assets/teo-simox-names.pdf))
- https://github.com/roboticslab-uc3m/teo-simox-models/blob/develop/doc/teo-simox-names.pdf ([permalink](https://github.com/roboticslab-uc3m/teo-simox-models/blob/75ab28ffd8dfaba33f77a073b4a2eee9be3218d1/doc/teo-simox-names.pdf))

AlvaroMartinezR, 2018-02-15:
Ups, my mistake, I forgot them :sweat_smile:.
Which one do you prefer that I delete? @jgvictores

AlvaroMartinezR, 2018-02-15:
Duplicates removed from https://github.com/roboticslab-uc3m/teo-simox-models/blob/develop/doc @jgvictores

jgvictores, 2018-02-15:
Thanks! I've added a link to the remaining file at https://github.com/roboticslab-uc3m/teo-simox-models/commit/536c5829fe1cedb1dad363ec06789baf796d22da

jgvictores, 2018-02-15:
Also did: https://github.com/roboticslab-uc3m/teo-developer-manual/commit/5d2ddb48c38585fb8e0e13b65b92c40754603a15

jgvictores, 2018-02-15:
Add visibility to dynamic info at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/5ca1cae47f81b91cb3149f3f639c848e56323731.

jgvictores, 2018-02-15:
Started #37 to track unifying link names.

jgvictores, 2018-02-15:
Been moving things around. We have https://github.com/roboticslab-uc3m/teo-developer-manual/blob/7d7f7216ecc2e558c65314d4862a3c90cc10a625/appendix/b-teo-dynamic-information.md but I'd also port https://github.com/roboticslab-uc3m/teo-simox-models/blob/551e1dc678eceea9a3ba416d7f584827a244df62/doc/teo-center-of-masses.csv to somewhere in https://github.com/roboticslab-uc3m/teo-developer-manual

jgvictores, 2018-12-12:
Blocked by #38

jgvictores, 2019-05-07:
Just for tracking:
- https://github.com/roboticslab-uc3m/teo-configuration-files/commit/f355beb437221ec9afb9a12dfcacd2b3b06cca6b was reverted at https://github.com/roboticslab-uc3m/teo-configuration-files/commit/ed94eaecbea1bee7af357d660f0b17e3a91ca889

jgvictores, 2019-05-07:
Also see: https://github.com/roboticslab-uc3m/teo-configuration-files/issues/17

jgvictores, 2019-05-08:
No longer blocked by https://github.com/roboticslab-uc3m/teo-main/issues/38

jgvictores, 2019-10-01:
Cool commit: https://github.com/roboticslab-uc3m/teo-developer-manual/commit/7def7696db5f6044710e2f9794c4e10991f6a178

imontesino, 2019-10-03:
Added the absolute values of the centers of masses in https://github.com/roboticslab-uc3m/teo-developer-manual/commit/ce9af815c7addfb406bdada0fd57e5e76baf8aa1
with respect to the right foot contact with the gound
[image]

jgvictores, 2019-10-03:
Cool!

- Correctly follows CAN ID numbering, currently: https://github.com/roboticslab-uc3m/teo-developer-manual/blob/cb372f68114a33151a0081fabc58c7bde65ed040/fig/Joints_(can).png
- Note that rotations are all positive since https://github.com/roboticslab-uc3m/teo-main/issues/39, as seen at https://github.com/roboticslab-uc3m/teo-developer-manual/blob/cb372f68114a33151a0081fabc58c7bde65ed040/fig/Directions.png