roboticslab-uc3m/teo-main #17: Review dynamic models (masses, COGs and inertias)

PeterBowman, 2017-08-16:
_From @jgvictores on March 7, 2015 8:35_

Review dynamic models (masses, COGs and inertias) currently in [teo-configuration-files/share/kinematics](https://github.com/roboticslab-uc3m/teo-configuration-files/tree/98bbc4aa181e21c2242dc915704432397534cab9/share/kinematics).

*Updated on 16/08/17 by PeterBowman.*

_Copied from original issue: roboticslab-uc3m/kinematics-dynamics#6_

PeterBowman, 2017-08-16:
_From @jgvictores on March 13, 2015 13:31_

Correct left and right arm kinematics and masses at [`037752e`](https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/037752e8d5e120e7fea1b0f77cdd17f7757240e9). COGs are approximate, and inertias are depreciated.

jgvictores, 2017-11-08:
Duplicate, unifying with #16.

closed 2022-06-04 by PeterBowman