roboticslab-uc3m/teo-main #18: Denavit Hartenberg revision for legs

PeterBowman, 2017-08-16:
_From @munozyanez on March 9, 2017 14:17_

Recent merge (https://github.com/roboticslab-uc3m/kinematics-dynamics/pull/83) has uploaded a revised DH parameters configuration in the following folder:
- https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/develop/share/kinematics
To be more specific, files 
- https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/share/kinematics/leftLegKinematics.ini
- https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/share/kinematics/rightLegKinematics.ini
Have changed according  DH Parameter Table published in http://robots.uc3m.es/index.php/TEO_Diagrams .

_Copied from original issue: roboticslab-uc3m/kinematics-dynamics#87_

PeterBowman, 2017-08-16:
_From @jgvictores on March 9, 2017 14:18_

@munozyanez Thanks a lot for this info. It will be very helpful in the future, and also in the present debugging issues such as #86.

PeterBowman, 2017-08-16:
_From @munozyanez on March 9, 2017 14:25_

By the way, I would like to propose to put all this files and ( https://github.com/roboticslab-uc3m/teo-openrave-models ) under specific teo-config repo. I mean, following the idea of teo-openrave-models, put all configuration files together can be useful for code cleanliness, and reuse with other robots.

PeterBowman, 2017-08-16:
_From @jgvictores on March 9, 2017 14:31_

> By the way, I would like to propose to put all this files and ( https://github.com/roboticslab-uc3m/teo-openrave-models ) under specific teo-config repo. I mean, following the idea of [...]

Yes, this is definitely a good idea. The only consideration is putting config files directly in teo-main, or in a new teo-config or teo-configuration-files etc, which will then be YCM'ed into teo-main. I'll be working on this tomorrow and will probably take a look at how it works directly in teo-main, without discarding the option of a new repo.

PeterBowman, 2017-08-16:
_From @munozyanez on March 24, 2017 16:16_

I can do it, but I am very busy at the moment. I can take if this is a long term assignment.

PeterBowman, 2017-08-16:
_From @munozyanez on March 24, 2017 16:56_

As you asked, the minimum knee angle for static gait for the robot is asin(0.1285/0.845) = 8.7 deg.
This angle sets the foot just under the root point. Angle must be a bit bigger.

PeterBowman, 2017-08-16:
DH parameters now go to [/share/kinematics](https://github.com/roboticslab-uc3m/teo-configuration-files/tree/9f08cf2e0a121ab0d714bea6cdb8ed295de9593c/share/kinematics) at our new repository https://github.com/roboticslab-uc3m/teo-configuration-files.

munozyanez, 2017-08-16:
Great! Fully agree with that approach.

jgvictores, 2017-12-04:
@munozyanez Are we happy with the leg kinematic model as of https://github.com/roboticslab-uc3m/teo-configuration-files/commit/3054b2e44dc66bca33d6f33f3671cfb62e7a0203 (current `develop` to close this issue?

munozyanez, 2017-12-04:
Yes, it is working ok afaik. Closing.

closed 2017-12-04 by munozyanez

jgvictores, 2017-12-04:
@munozyanez Thanks!