roboticslab-uc3m/teo-main #19: How to do a grasping hand for Teo

jgvictores, 2017-10-28:
_From @jgvictores on October 4, 2017 15:26_

_From @rsantos88 on October 4, 2017 8:33_

To advance in the correct way with [grasping in openrave](http://openrave.org/docs/0.6.6/openravepy/databases.grasping/), we need to make the actual Teo hand able to open and close like the [barrett hand](https://github.com/rdiankov/openrave/blob/master/src/robots/barretthand.kinbody.xml).
Which are the steps to follow?

_Copied from original issue: roboticslab-uc3m/openrave-yarp-plugins#37_

_Copied from original issue: roboticslab-uc3m/questions-and-answers#31_

jgvictores, 2017-10-28:
_From @rsantos88 on October 4, 2017 8:55_

First, I think we need to separate the fingers in three different 3D files..

[image]

or.. maybe 6 files? (2 parts per finger..?)

jgvictores, 2017-10-28:
Blocked by: https://github.com/roboticslab-uc3m/teo-openrave-models/issues/9

rsantos88, 2017-12-12:
A grasping hand was added to the robot

jgvictores, 2018-04-04:
Old issue closed, now blocked by https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/86

jgvictores, 2019-11-20:
Closed https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/86, now working on https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/97

PeterBowman, 2021-11-09:
The referenced issues have been closed, and now we also have a fully actuated gripper in the Gazebo models thanks to https://github.com/roboticslab-uc3m/teo-gazebo-models/pull/4. @jgvictores what are the next steps?

jgvictores, 2021-11-09:
Original author was not me, I'm just mentioned due to the old issue move mechanism.

I'd close! :)

jgvictores, 2021-11-09:
ps: oh yes, we did all this. closing!

closed 2021-11-09 by jgvictores