roboticslab-uc3m/teo-main #21: Dynamic Information about TEO

jgvictores, 2017-11-08:
_From @AlvaroMartinezR on November 8, 2017 12:15_

I have been working with SolidWorks and other CAE programs in order to develop an accurate dynamic model for TEO, including masses, inertias, CoM. It is not yet finished, but I wonder in which repository this kind of information (probably a .doc or a .pdf file) could be placed into.

_Copied from original issue: roboticslab-uc3m/teo-openrave-models#11_

jgvictores, 2017-11-08:
_From @PeterBowman on November 8, 2017 12:18_

These properties are parameters to the cartesian solver and usually go to [teo-configuration-files](https://github.com/roboticslab-uc3m/teo-configuration-files). Take a look at [share/kinematics/](https://github.com/roboticslab-uc3m/teo-configuration-files/tree/e52590665a54cfbf33863f6e6264bdcc1df154a1/share/kinematics).

Probably we should move this to roboticslab-uc3m/teo-main#17.

jgvictores, 2017-11-08:
Duplicate, unifying with #16.

AlvaroMartinezR, 2017-11-09:
yo al menos no saco un 6 en

closed 2022-06-04 by PeterBowman