roboticslab-uc3m/teo-main #22: A couple of problems related to Teo movements

jgvictores, 2017-11-14:
_From @rsantos88 on November 14, 2017 17:28_

There are a couple of problems related to Teo movements and it can ruin a demonstration (leading to reset it several times... :sweat:)
* The first occurs randomly, when the program sends  a sequence of movements, for example salute, composed by four movements: lift arm up, axial arm turn in, axial arm turn out, move arm down.
We can see that this movement is not executed correctly in certain occasions. The arm finishes the movements early, running all at once and not sequentially. This can produce a freeze of all the demo, because the movement has not finished correctly all the sequence. Can it be a problem with checkCanBus() function? Why does it works well sometime and sometimes not?
* Other problem observed is, when Teo makes a movement (I would highlight **axial movements**), sometimes is not finished and the joint starts to shake (like a parkinson). This may take a while (shaking horribly) until the movement finally ends.

I don't know where I can post this problems, because the first can be a software problem, but the second it can be a tunning motors problem ¿? I don't know
I need some ideas to solve it.

_Copied from original issue: roboticslab-uc3m/questions-and-answers#34_

jgvictores, 2017-11-14:
> - The first occurs randomly, when the program sends a sequence of movements, for example salute, composed by four movements: lift arm up, axial arm turn in, axial arm turn out, move arm down.
We can see that this movement is not executed correctly in certain occasions. The arm finishes the movements early, running all at once and not sequentially. This can produce a freeze of all the demo, because the movement has not finished correctly all the sequence. Can it be a problem with checkCanBus() function? Why does it works well sometime and sometimes not?

@rsantos88 I doubt this is related with `checkCanBus()`. It should be more related with `checkMotionDone` and delays. Could you link some example code where this happens?

> - Other problem observed is, when Teo makes a movement (I would highlight axial movements), sometimes is not finished and the joint starts to shake (like a parkinson). This may take a while (shaking horribly) until the movement finally ends.

That definitely looks like a driver tuning issue. Not that it has a simple solution, we're joint-level PID controllers for changing loads with coupled joints and links (a.k.a. complex multi-body dynamics).

rsantos88, 2017-11-15:
> @rsantos88 I doubt this is related with checkCanBus(). It should be more related with checkMotionDone and delays. Could you link some example code where this happens?

Sorry @jgvictores . It was a mistake. I meant checkMotionDone(). You have examples in teo-self-presentation demo, with [salute](https://github.com/roboticslab-uc3m/teo-self-presentation/blob/master/programs/BodyExecution/BodyExecution.cpp#L201-L226) movement here. This state call to [jointsMoveAndWait](https://github.com/roboticslab-uc3m/teo-self-presentation/blob/master/programs/BodyExecution/BodyExecution.cpp#L117-L166) which makes use of the [checkMotionDone()](https://github.com/roboticslab-uc3m/teo-self-presentation/blob/master/programs/BodyExecution/BodyExecution.cpp#L148) function. Don't be scared if you see that part of the code regarding the left arm has been [commented](https://github.com/roboticslab-uc3m/teo-self-presentation/blob/master/programs/BodyExecution/BodyExecution.cpp#L151-L157). This is a temporal solution for the problems caused by the change of piece that is being made for the rotation of the left wrist (like [this](https://github.com/roboticslab-uc3m/teo-hardware-manual/issues/20)).
Other example, can be in follow-me demostration, with [salute](https://github.com/roboticslab-uc3m/follow-me/blob/develop/programs/followMeArmExecution/FollowMeArmExecution.cpp#L190-L220) movement or the [arm swimming](https://github.com/roboticslab-uc3m/follow-me/blob/develop/programs/followMeArmExecution/FollowMeArmExecution.cpp#L163-L188). In the arm swimming case, you can see that the arm doesn't finish its movement or it stays frozen for a while until it regains the swinging motion.

jgvictores, 2017-11-15:
1. **General programming recommendation:** I'd recommend introducing a small delay *before* the `checkMotionDone()` (as commented somewhere among YARP issues).

1. **Note on movement and sincronization of movements:** Note that in the cited functions, **velocity** (not **duration**) is set to the same for all joints. This that the joint that has to move less arrives first, the second smallest increment reaches second, and so on.

1. **Note on arm swinning:** Note that arm swinging is pretty hackish. Cycle time should be adapted to the velocity. I would not point to it as a reference at all.

1. **Note on GitHub issues:** Please point to the specific commit (press `y` to get it on you URL, it works a bit more as a permanent link than without it).

Please provide feedback after implementing (1) and taking (2) and (3) into account. Thanks!

rsantos88, 2017-12-12:
Blocked by https://github.com/roboticslab-uc3m/teo-hardware-manual/issues/23

jgvictores, 2018-01-26:
Old issue that was blocking is now https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/2 which has also been closed. Unblocking!

rsantos88, 2018-01-31:
Issue roboticslab-uc3m/teo-hardware-issues#2  was resolved.
This is unblocked now

rsantos88, 2018-01-31:
I'm having problems with this:
> Other problem observed is, when Teo makes a movement (I would highlight axial movements), sometimes is not finished and the joint starts to shake (like a parkinson). This may take a while (shaking horribly) until the movement finally ends.

[image]

Could you fix it @jmgarciah? :innocent:

rsantos88, 2018-02-02:
> General programming recommendation: I'd recommend introducing a small delay before the checkMotionDone() (as commented somewhere among YARP issues).

That point is fixed [here](https://github.com/roboticslab-uc3m/teo-self-presentation/blob/develop/programs/BodyExecution/BodyExecution.cpp#L220-L236) and probably it works better now. I've to do more tests. 
I am pending of the second point. I've opened a issue https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/14

rsantos88, 2018-02-12:
No more blocked.
I'll close this issue until the problems reappear again ;)

closed 2018-02-12 by rsantos88