roboticslab-uc3m/teo-main #23: Changes of forearm and its consequences

rsantos88, 2017-11-15:
Respect with the last changes of [forearm piece](https://github.com/roboticslab-uc3m/teo-hardware-manual/issues/20), it remains to make all the changes regarding the new kinematics, diagrams, model of Teo, etc ...

jgvictores, 2017-11-15:
Okay, please remember the current [workflow-for-assets in teo-software-manual/](https://github.com/roboticslab-uc3m/teo-software-manual/tree/e88693d314a04309cfbacc09b9bb6636b176e4b4#workflow-for-assets).

jgvictores, 2018-02-15:
Blocks #38 among others.

jgvictores, 2019-05-06:
@rsantos88 This affects https://github.com/roboticslab-uc3m/teo-main/issues/38

Are the current values at https://github.com/roboticslab-uc3m/teo-developer-manual/blob/cabf02d5438e2735a4983b82f9ff0c6fca2fe6cd/csv/lengths.csv correct (via diagram at https://github.com/roboticslab-uc3m/teo-developer-manual/blob/cabf02d5438e2735a4983b82f9ff0c6fca2fe6cd/fig/Lengths.png)?

rsantos88, 2019-05-07:
updated via https://github.com/roboticslab-uc3m/teo-developer-manual/pull/58

rsantos88, 2019-05-07:
also updated kinematic files via https://github.com/roboticslab-uc3m/teo-configuration-files/pull/22

jgvictores, 2019-05-07:
Done by @rsantos88 via:
- https://github.com/roboticslab-uc3m/teo-developer-manual/pull/58
- https://github.com/roboticslab-uc3m/teo-configuration-files/pull/22

closed 2019-05-07 by jgvictores