roboticslab-uc3m/teo-main #26: Try hc-driver-4.9 on TEO

jgvictores, 2017-11-15:
Try hc-driver-4.9 on TEO.

If https://github.com/munozyanez/hc-driver-4.9 works, this may affect the Distro we use, propagating to https://github.com/roboticslab-uc3m/yarp-devices/issues/111 and even https://github.com/roboticslab-uc3m/QA/issues/17.

This issue originates from https://github.com/roboticslab-uc3m/yarp-devices/issues/118 but is specific to TEO.

jgvictores, 2017-11-21:
Adding `priority: high` given the number of issues that are blocked by this issue and related https://github.com/roboticslab-uc3m/yarp-devices/issues/118.

rsantos88, 2017-11-30:
this issue is blocked by @jmgarciah and @smcdiaz 
They are doing tests about push-recovery

jgvictores, 2017-11-30:
@rsantos88 Thanks for keeping us updated on this topic. It is extremely important, as this issue is keeping many other issues `blocked` too.

jgvictores, 2017-12-07:
> this issue is blocked by @jmgarciah and @smcdiaz  They are doing tests about push-recovery
Does this refer to https://github.com/roboticslab-uc3m/TEO_push repository, or a different one (to put an `announcement` issue pointing somewhere)?

AlvaroMartinezR, 2017-12-19:
@rsantos88 told me that he is preparing to install in a partition of TEO Ubuntu 16.04. I think is a good option but I want to point out that the ArmarX framework is recommended to be used in 14.04. However thinking about installing ArmarX in TEO is a long shot: first, the idea is to create a simulation of walking in a non-connected PC. And, if achieved, translate to TEO. 
To conclude, my point is that my consideration should be less pondered over others for 2 facts:

- This is a very long goal.

- I suppose that something that works on 14.04 should work on 16.04.

jgvictores, 2017-12-19:
Most of our Travis CI are with Trusty (14.04.xx), so I guess it would be reasonable to go for Trusty rather than Xenial (16.04.xx).
Since this is a kind of organization-wide issue, I'd even consider another place where we're stuck with Trusty: TIAGo (all for ROS Indigo->Trusty).
So, Trusty kind of seems like the way to go, leaving Xenial as an option of a personal preference. Hopefully ArmarX, TIAGo, and Travis will all end up moving to Xenial more or less at the same pace.

@PeterBowman What is your opinion?

PeterBowman, 2017-12-19:
> Since this is a kind of organization-wide issue (...)

Is it? We have a robot-wide discussion at roboticslab-uc3m/QA#20, but this one is perhaps more closely related to TEO (and, by extension, to ASIBOT with regard to HicoCAN driver support). Let's break down a few points:

* According to [this blog entry](https://blog.travis-ci.com/2017-12-01-new-update-schedule-for-linux-build-images), Xenial support in Travis CI is just around the corner.
* ArmarX: I can't tell much about it. The package repository only lists Ubuntu 14.04, but have you tried [installing from sources](https://armarx.humanoids.kit.edu/ArmarXDoc-ArmarX-Installation-Manual.html) in Xenial?
* I can't see the connection between TIAGo and this issue - how is it related to the OS aboard TEO?

One of the main concerns for moving away from Debian 6.0 to a modern SO was the C++11 support requirement dictated by YARP. This is fulfilled by Trusty, so I guess we can give it a try and defer the final upgrade to a later stage. However, please keep in mind that we are already behind YARP's expectations for supported distros: http://wiki.icub.org/wiki/YARP_Supported_Distributions#Supported_releases. There is some sort of buffer zone in our org thanks to roboticslab-uc3m/QA#17, but we could run into problems sooner or later.

rsantos88, 2017-12-20:
Ok, so I suppose we are all in agreement with using Ubuntu 14.04 LTS for all of TEO PCs

jgvictores, 2017-12-20:
I think it fulfills our short-term objective of trying the driver, and also enables simple tests with ArmarX, which is good. 

Both @David-Estevez an I have introduced features in YARP that are on `devel` which will require Ubuntu 16.04. Just to think about the future, do we physically have space for (worst case scenario) 3 distros?

AlvaroMartinezR, 2017-12-20:
> ArmarX: I can't tell much about it. The package repository only lists Ubuntu 14.04, but have you tried installing from sources in Xenial?

No, I haven't. The computer I work with has no enough space to think about adding another partition just to test. 

But as I said, for TEO my opinion should not be regarded as compulsory for my work, I was talking in a very long term.

rsantos88, 2017-12-20:
Yes, we have enough space. More in detail, for the debian 6.0 backup, the partition is 20gb and we have approximately 100gb for the rest of distros. So, do you want to have installed in each PC.. ubuntu 14.04 LTS, ubuntu 16.04 LTS and the Debian backup?

rsantos88, 2017-12-20:
ok, as I told to @jgvictores , I'll install ubuntu 14.04 LTS and I'll leave a free partition with 50gb for a future distro (ubuntu 16.04)

jgvictores, 2017-12-20:
@rsantos88 Perfect, thanks!!

rsantos88, 2017-12-21:
Ok, now we have teo-manipulation and teo-locomotion PCs with this grub menu:

[image]
By default, it will start with the old distribution (backup 23/10/2017 with Debian 6.0).
If you select `Ubuntu`, you will start a clean distribution of `Ubuntu 14.04 LTS` located in a clean partition of 54gb.
I haven't done this for now with teo-head PC because it has 'ubuntu 15.04`

jgvictores, 2017-12-21:
Thank you so much! Looks very good!!

jgvictores, 2018-01-17:
> this issue is blocked by @jmgarciah and @smcdiaz They are doing tests about push-recovery

Spoke with @jmgarciah and @smcdiaz, no longer blocked!

rsantos88, 2018-01-18:
Perfect. If you need to install something additional (libraries, repositories, etc) in that distribution, tell me.

jgvictores, 2018-01-18:
All we need to try is launchManipulation/launchLocomotion basic joint control.

rsantos88, 2018-01-19:
consider this https://github.com/roboticslab-uc3m/installation-guides/issues/34

jgvictores, 2018-01-20:
Installing the new hico driver should be very similar to that of the old driver, probably just changing some paths: https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-hcanpci.md#install-hcanpci-kernel-module-debian-60

rsantos88, 2018-01-23:
> All we need to try is launchManipulation/launchLocomotion basic joint control.

Now `launchManipulation` is running perfectly with Ubuntu 14.04 !! :thumbsup:
I will update locomotion computer and document it with the new drivers :)

jgvictores, 2018-01-23:
Amazing!! 😎😎😎

Thanks a lot!!!

rsantos88, 2018-01-23:
Do you consider the possibility of storing the drivers in another repository?

rsantos88, 2018-01-24:
`launchLocomotion` now is ready too for Ubuntu 14.04
The documentation has been updated with the new drivers for Ubuntu distro [here](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-hcanpci.md#install-hcanpci-kernel-module-debian-60)

rsantos88, 2018-01-25:
maybe we should update [this](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/doc/yarp-devices-install.md#os-requirements) table with CanBusControlboard and CanBusHico working

jgvictores, 2018-01-28:
Regarding the table, note that we have https://github.com/roboticslab-uc3m/yarp-devices/issues/161 open.

rsantos88, 2018-01-31:
ToDo: test jr3 drivers with Ubuntu 14.04

rsantos88, 2018-01-31:
Closing issue! The rest of tasks for Distro update will be tracked on #32

closed 2018-01-31 by rsantos88