roboticslab-uc3m/teo-main #32: Migration from Debian 6.0 to Ubuntu 14.04 on TEO

jgvictores, 2018-01-31:
Migration from Debian 6.0 to Ubuntu 14.04 on TEO

ToDo: 
- [ ]  jr3 drivers with Ubuntu 14.04
- [ ]  update [this](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/doc/yarp-devices-install.md#os-requirements) table using the agreed format (csv)

jgvictores, 2018-01-31:
> jr3 drivers with Ubuntu 14.04

Blocked by https://github.com/roboticslab-uc3m/LoliRepo/issues/1

jgvictores, 2018-01-31:
Jr3 on Ubuntu still has issues... On Debian `lsmod` we have:
```
jr3pci_driver          23698  0 
jr3_pci                 6086  0 
comedi                 24352  1 jr3_pci
```

jgvictores, 2018-01-31:
JR3 working via https://github.com/roboticslab-uc3m/jr3pci-linux/commit/3cb1a29cf3782b3d3808d3e8f9c2d65a91499879 (current `develop` of https://github.com/roboticslab-uc3m/jr3pci-linux )!!

jgvictores, 2018-01-31:
- No longer blocked by https://github.com/roboticslab-uc3m/LoliRepo/issues/1.
- WIP at https://github.com/roboticslab-uc3m/installation-guides/issues/35

jgvictores, 2018-01-31:
- https://github.com/roboticslab-uc3m/installation-guides/issues/35 closed.

jgvictores, 2018-01-31:
- JR3 working on Ubuntu 14.04 and documented at https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-jr3.md
- XSENS working on Ubuntu 14.04 and documented at https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-xsens.md

jgvictores, 2018-01-31:
Updated links to these docs on https://github.com/roboticslab-uc3m/teo-software-manual/blob/fb1c7b73235bfacab411b4fe9d344b2be2d0c4e5/appendix/b-teo-network-information.md

jgvictores, 2018-01-31:
> Install daemontools as described here
```
The YARP_CONF variable is deprecated and it is no longer used. Please check the documentation for yarp::os::ResourceFinder::getConfigHome() warning: YARP name server(s) not configured, ports will be anonymous warning: check your namespace and settings with 'yarp detect'
```

rsantos88, 2018-02-01:
From [here](https://github.com/robotology/yarp/blob/9712fdec65a676247d9a297928ff5f4ad8286901/src/libYARP_OS/src/NameConfig.cpp#L80-L94) we see `YARP_CONF` is deprecated, and from the called `ResourceFinder::getConfigHome()`, and from [its documentation](https://github.com/robotology/yarp/blob/9a395f6e41e01d8d350b6a4e5e2b2f6c6f913b74/src/libYARP_OS/include/yarp/os/ResourceFinder.h#L359-L376) we can see that `YARP_CONFIG_HOME` is now used instead.

Tested and working! :tada::tada::tada:

rsantos88, 2018-02-01:
Starting documentation!
Shoud create GitHub releases, before erasing all Debian 6.0 documentation, at least on https://github.com/roboticslab-uc3m/yarp-devices

jgvictores, 2018-02-02:
Created https://github.com/roboticslab-uc3m/yarp-devices/releases/tag/old-debian-6.0

jgvictores, 2018-02-02:
Created https://github.com/roboticslab-uc3m/installation-guides/releases/tag/old-debian-6.0

rsantos88, 2018-02-06:
>  update [this](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/doc/yarp-devices-install.md#os-requirements) table using the agreed format (csv)

Here it is: https://github.com/roboticslab-uc3m/teo-main/blob/issue%2332/doc/os-requirements.csv

rsantos88, 2018-02-06:
pull request:  #34

rsantos88, 2018-02-06:
Do you want to convert [this](https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/doc/yarp-devices-install.md#specific-dependencies) table to .cvs too, don't you?

jgvictores, 2018-02-06:
In fact most work done at https://github.com/David-Estevez/yarp-devices/tree/fix-161-migrate-tables, check issue https://github.com/roboticslab-uc3m/yarp-devices/issues/161.
The basic idea is to conclude https://github.com/roboticslab-uc3m/yarp-devices/issues/161 but taking into account that all Debian 6.0 information can be removed.

jgvictores, 2018-02-06:
Also careful with https://github.com/roboticslab-uc3m/yarp-devices/commit/98ec0efa63423406cb94fe49b76eaee01d4e8995 (fix-165-find-phidget)

jgvictores, 2018-02-07:
Added "Change boot order: make Ubuntu default." to description.

rsantos88, 2018-02-08:
> Added "Change boot order: make Ubuntu default." to description.

Done in both computers. Now you can see the next screen when grub starts, ease to recognize:
(Default time to choose: 5s)

[image]

jgvictores, 2018-02-08:
> Do you want to convert this table to .csv too, don't you?

Great. Basically missing merge of the branches that involve this file.

jgvictores, 2018-02-08:
Finished documentation, which was final step, at ffce5d9918cf5b8c6a81a2dea1d447edde772b15.

closed 2018-02-08 by rsantos88