roboticslab-uc3m/teo-main #35: Not all drivers act in torque mode with launchManipulation + gcmp

jgvictores, 2018-02-08:
Using:
- launchManipulation which uses https://github.com/roboticslab-uc3m/yarp-devices
- gcmp from BasicCartesianControl from https://github.com/roboticslab-uc3m/kinematics-dynamics

launchManipulation can be launched, but having issues when used with stated `[gcmp]`: torque mode is set, but driver 1 of rightArm seems to be in position control (note that kin-dyn stat and yarp-dev [get] [icmd] [cmds] indicate all okay). Sending kin-dyn `[stop]` then `[gcmp]` again makes it worse: despite that kin-dyn stat and yarp-dev [get] [icmd] [cmds] indicate all okay, all act in practice as if position mode.

jgvictores, 2018-02-08:
Blocks https://github.com/roboticslab-uc3m/xgnitive/issues/52

PeterBowman, 2018-02-08:
Possibly related: roboticslab-uc3m/yarp-devices#127.

jgvictores, 2018-02-08:
> Possibly related: roboticslab-uc3m/yarp-devices#127.

Yes, looks totally like that. Attempting to: pos -> torq -> pos -> torq seems not doable at all.

jgvictores, 2018-02-08:
Hack for now will possibly be enabling and checking each motor driver manually, before launching `gcmp`... :-(

jgvictores, 2018-02-09:
Progress will be tracked at https://github.com/roboticslab-uc3m/yarp-devices/issues/170 ([TechnosoftIpos] Problems changing mode)

closed 2022-06-04 by PeterBowman