roboticslab-uc3m/teo-main #36: Error compilation on manip-waiter repo

jlpesoto, 2018-02-08:
there is an error compilation on manip-waiter repo (branch: jr3) related with ICartesianSolver.h

In file included from /home/teo/repos/manip-waiter/programs/jr3WristControl/InCvPort.cpp:3:0:
/home/teo/repos/manip-waiter/programs/jr3WristControl/InCvPort.hpp:14:30: fatal error: ICartesianSolver.h: No such file or directory
compilation terminated.
programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/build.make:62: recipe for target 'programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/InCvPort.cpp.o' failed
make[2]: *** [programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/InCvPort.cpp.o] Error 1
CMakeFiles/Makefile2:332: recipe for target 'programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/all' failed
make[1]: *** [programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

PeterBowman, 2018-02-08:
> ICartesianSolver.h

This header is shipped with the [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics/) repo. Please follow the installation guides, you should end up with ICartesianSolver.h in /usr/local/include.

jmgarciah, 2018-02-08:
A new error appears related with ColorDebug.hpp:

/home/teo/repos/manip-waiter/programs/jr3WristControl/InCvPort.hpp:16:26: fatal error: ColorDebug.hpp: No such file or directory
compilation terminated.
programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/build.make:62: recipe for target 'programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/InCvPort.cpp.o' failed
make[2]: *** [programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/InCvPort.cpp.o] Error 1
CMakeFiles/Makefile2:329: recipe for target 'programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/all' failed
make[1]: *** [programs/jr3WristControl/CMakeFiles/jr3WristControl.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

jgvictores, 2018-02-08:
This issue was moved to roboticslab-uc3m/manip-waiter#1

closed 2022-06-04 by PeterBowman