roboticslab-uc3m/teo-main #37: Review and Unify TEO link names

jgvictores, 2018-02-15:
Review and Unify TEO link names, in parallel to  https://github.com/roboticslab-uc3m/teo-main/issues/16 (dynamic info), etc.

jgvictores, 2018-02-15:
Note: would like to change teo-simox-names.pdf -> teo-link-names.pdf or similar.
Must update the following current items after that:
- https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/appendix/a-teo-diagrams.md
- https://github.com/roboticslab-uc3m/teo-simox-models/blob/develop/README.md

AlvaroMartinezR, 2018-03-09:
@jgvictores I am not sure fi you want to change the names I used in simox to other or the other way around, change the simox ones.

jgvictores, 2018-03-11:
Simox names are good! 
Unless there is some other naming for the links, I think it would be cool to adopt the ones you have on the PDF everywhere (thus could drop the simox from name, as becomes generic info).

AlvaroMartinezR, 2018-03-12:
Ok, now I understand. I know that the ones used in openrave are something like r10.001.

jgvictores, 2018-03-12:
Will probably be renaming and moving things around today, have presentation on TEO tomorrow evening! :-)

PeterBowman, 2019-06-26:
Note: "saggital, frontal, axial..." might be misplaced, e.g. current saggital leg lniks are actually frontal links according to https://en.wikipedia.org/wiki/Anatomical_plane.

PeterBowman, 2019-11-04:
Note: review and unify OpenRAVE link names, too.

jgvictores, 2019-11-04:
> Note: review and unify OpenRAVE joint names, too.

Updated issue description. May have added some overkill: for now our `.ini`'s do not contain this sort of info.

jgvictores, 2019-11-07:
ASWB, we'll name links as previous joints, in a DH fashion.

rsantos88, 2019-11-08:
In order to help with this issue, and to be sure that I've understood it, do you want to replace, for example `q10` [here](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/develop/openrave/teo/teo_rightArm.kinbody.xml#L20 ) by `FrontalRightShoulder` or by the label `FRS15` following [this](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/csv/motores-motores.csv) table? Is it right?

jgvictores, 2019-11-09:
Hi! The table is the correct one, and we'd use the first column, e.g. `FrontalRightShoulder ` as you said. 

However, [`q10`](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/a27e6b9e8708e52b198ebea80c6d3579b419978d/openrave/teo/teo_rightArm.kinbody.xml#L20) would be a **joint**, to be done at https://github.com/roboticslab-uc3m/teo-main/issues/44. <- We can start with this issue first! :smile:

This issue refers to **links**, with names such as [`r10`](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/a27e6b9e8708e52b198ebea80c6d3579b419978d/openrave/teo/teo_rightArm.kinbody.xml#L7). This can be slightly more tedious. Things to be taken into account:
- Possibility of renaming files within [teo/models](https://github.com/roboticslab-uc3m/teo-openrave-models/tree/a27e6b9e8708e52b198ebea80c6d3579b419978d/openrave/teo/models), best via `git mv`.
- Updating in all `.xml` files, including unexpected places such as [here](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/a27e6b9e8708e52b198ebea80c6d3579b419978d/openrave/teo/teo_lacqueyFetch.robot.xml#L10-L15).

jgvictores, 2019-11-12:
Regarding the OpenRAVE model, since https://github.com/roboticslab-uc3m/teo-openrave-models/pull/35 has been merged, I guess this could be done (warning! pretty tedious ^^).

rsantos88, 2019-11-12:
> - [x] Openrave models: https://github.com/roboticslab-uc3m/teo-openrave-models

Done at https://github.com/roboticslab-uc3m/teo-openrave-models/pull/36 according to https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/fig/teo-link-names.pdf