roboticslab-uc3m/teo-main #39: Review and unify Joint Directions of Rotation

jgvictores, 2018-02-15:
Review Joint Directions of Rotation https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/appendix/a-teo-diagrams.md#joint-directions-of-rotation then unify

rsantos88, 2018-02-19:
I found that some directions of rotation are wrong.
Do you want to correct the diagram or the direction in the real robot?

rsantos88, 2018-02-19:
The signs are ok but, @jmgarciah and me think that the arrows can generate confusin in the Y axis

jmgarciah, 2018-02-20:
Talking with @smcdiaz and seeing the TEO's hardware, we have found a possible solution. Related with the two saggital motor of the legs, the wires of the gear are different from the rest. 

@jgvictores have given me permission to change the order of the wires and have the same turning sense that the diagram.

jmgarciah, 2018-02-21:
I have found an image that could be insteresting for a new TEO diagram. And finally understand the joint direction.

[image]

jgvictores, 2018-02-21:
Looks nice, could be done with TEO.

I would try to avoid the RPY terminology (Roll is Z in some fields) and stick to ours.

jmgarciah, 2018-02-21:
I totally agree. It is only an idea. So please, take it as you need

RaulFdzbis, 2018-02-21:
In my opinion, i think that overlapping of the robot model over the axis, make that image a little overload, and can be confusing.

Maybe a good solution could be to use the simplified model that is used now with the Teo Joint Directions of Rotation but with the perspective in that image, for an easier understanding.

jgvictores, 2018-02-21:
If anyone could take a picture of TEO in a pose similar to NAO there (same perspective, but arms pointing down), I could give it a shot. Some alpha layer magic should do the trick. :-)

David-Estevez, 2018-02-21:
This is not a picture, but might be useful:
[image]

jmgarciah, 2018-02-21:
Something like @David-Estevez shows could be perfect

David-Estevez, 2018-02-21:
I can pose it and change the colors to whatever suits your diagrams, just tell me!

rsantos88, 2018-02-21:
@David-Estevez totally agree :+1:

jmgarciah, 2018-02-22:
@smcdiaz and me have finished to correct the saggital joint direction. But we have to check that everything is ok

rsantos88, 2018-02-22:
> @smcdiaz and me have finished to correct the saggital joint direction. But we have to check that everything is ok

I've checked the result with @jmgarciah and it's not working well. 
When I run `launchLocomotion` and check the value of sagital axis, the value is `-3.93673706054688`
I move this axis to homing position, setting position to 0, and the axis moves outside and when I get the position, it returns `-7.45166015625`
@jmgarciah propouse to @jgvictores or @smcdiaz to have a look at this problem because
he has no idea.

rsantos88, 2018-02-26:
Changed sign of CUI Absolute Encoder to read the correct joint directions of rotation.
See [here](https://github.com/roboticslab-uc3m/teo-configuration-files/pull/8) the changes. It has been tested and it works ok. Now the directions of rotation of the joint ID 5 and 12 correspond to the diagram.

Note: the axial rotation of the waist has not been corrected

rsantos88, 2018-03-07:
notes: 
* [PR](https://github.com/roboticslab-uc3m/teo-configuration-files/pull/8) is pending to be performed, but it has been changed on the robot to be use.
* After all the changes, I've corrected the home position of the legs (more accurate initial position)

rsantos88, 2018-03-07:
Other curious note observed: 
* When you run `launchLocomotion`, always the right sagittal leg moves arround 10º degrees out.
You can correct this doing the home position with `--homePoss`, but something causes the joint to move when the driver is initializing.

jmgarciah, 2018-03-08:
The legs' sagittal joints are already fixed. The only remaining joint is the axial waist. Fixing via hardware this joint is more complicate than via software, I think. It is neccesary to separe the body from the legs. I propose to change the values from .ini files. Please @jgvictores , help us.

jgvictores, 2018-03-08:
On my way!

jgvictores, 2018-03-08:
> It is neccesary to separe the body from the legs. I propose to change the values from .ini files

First diagrams, then update all related configuration files and robot models.

jgvictores, 2018-03-22:
The fact that there are negative `trs` in `launchLocomotion.ini` means that diagrams are incorrect.
I've opened https://github.com/roboticslab-uc3m/teo-configuration-files/issues/11 pointing at the situation.

rsantos88, 2018-03-22:
> The fact that there are negative trs in launchLocomotion.ini means that diagrams are incorrect.
> I've opened roboticslab-uc3m/teo-configuration-files#11 pointing at the situation.

https://github.com/roboticslab-uc3m/teo-configuration-files/issues/11 has been closed.

rsantos88, 2018-06-11:
The direction of rotation of the axial waist is opposite in the `.ini` config file and the diagram.
What we should to change? The diagram? The config file? or it doesn't matter which of them?

jgvictores, 2018-06-11:
Better change the .ini (diagrams also affect DH etc)

rsantos88, 2019-02-25:
Ok, I'm going to work on this issue:
I will use the rotation diagram as a reference, correcting those that have a negative direction and applying the change to the maximum and minimum limits:

[image]

I will use the example of the right elbow that @jgvictores  told me:

[image]

The result, taking into account the current limits of manipulation and locomotion would be [this](https://github.com/roboticslab-uc3m/teo-configuration-files/commit/5addf18fbf8fb467ca46c082a22920406c2d7f8c).

In my humble opinion, I think it would be quite interesting to reconsider the revision of the limits. In the arms a more detailed revision of the limits has already been made, correcting some errors. (https://github.com/roboticslab-uc3m/teo-main/issues/15#issuecomment-461091935)
I would work on these limits. Also, I would recommend checking the limits of the legs again with @jmgarciah , and apply the new changes of direction. I await your proposals :)

jgvictores, 2019-02-25:
@rsantos88 I've taken a look at https://github.com/roboticslab-uc3m/teo-configuration-files/commit/5addf18fbf8fb467ca46c082a22920406c2d7f8c and it seems amazing!

-To close this issue I'd be satisfied with real TEO's ini files and manual images, and open separate issues on simulation model repositories
- Note to self: develop small script to enable porting programs to be compatible with new criteria

Issues that will be unblocked:
- Joint limits: see my comment at https://github.com/roboticslab-uc3m/teo-main/issues/15#issuecomment-467021993
- Kinematic model and diagrams: https://github.com/roboticslab-uc3m/teo-main/issues/38
- Probably many more!

rsantos88, 2019-03-15:
> To close this issue I'd be satisfied with real TEO's ini files and manual images, and open separate issues on simulation model repositories

I've modified the images of directions. Review it please: https://github.com/roboticslab-uc3m/teo-developer-manual/commit/6238c4bc0d75b93c2e416fc87071aecf7e94cf61

> In my humble opinion, I think it would be quite interesting to reconsider the revision of the limits. In the arms a more detailed revision of the limits has already been made, correcting some errors. (https://github.com/roboticslab-uc3m/teo-main/issues/15#issuecomment-461091935)
I would work on these limits. Also, I would recommend checking the limits of the legs again with @jmgarciah , and apply the new changes of direction. I await your proposals :)

To continue with this issue, once again, I would recommend modifying and correcting the limits of the arms and legs and then the direction of rotation, are you agree with that?

jgvictores, 2019-03-15:
Looks great to me!

IMHO, you can close the issue, and we can proceed with:
- [ ] https://github.com/roboticslab-uc3m/teo-main/issues/15 (joint limits)
- [ ] Update simulated models (must put issue per repo)
- [ ] Update demos (must put issue per repo)
   - https://github.com/roboticslab-uc3m/waiter
   - https://github.com/roboticslab-uc3m/follow-me
   - https://github.com/roboticslab-uc3m/teo-self-presentation

rsantos88, 2019-03-19:
Here are the result of merging the new **joint limits of arms** with the new **direction of rotation** in "teo-configuration-files" repository: https://github.com/roboticslab-uc3m/teo-configuration-files/pull/18 
(please, check it in case something has escaped me)

Maybe we can keep it this way, without revising the limits of the legs again and consider that the previous measurement with @AlvaroMartinezR  was correct. What do you think? Would it be interesting to review the limits of the legs again before continuing?
If we continue without legs revision, the next steps would be:
- [ ] Accept this pull request
- [ ] Reflect the changes in the tables of **teo-developer-manual** repository

and...
>- [ ] Update simulated models (must put issue per repo)
>- [ ] Update demos (must put issue per repo)
>   - https://github.com/roboticslab-uc3m/waiter
>   - https://github.com/roboticslab-uc3m/follow-me
>   - https://github.com/roboticslab-uc3m/teo-self-presentation

jmgarciah, 2019-03-19:
I think you should check the whole robot, not just the arms. If we are going to modify the directions of the joints, we should not leave the work halfway.

rsantos88, 2019-03-21:
Updated: https://github.com/roboticslab-uc3m/teo-main/issues/39#issuecomment-474334416

jgvictores, 2019-03-27:
> * [x]  Accept pull request (`tr`s in `yarp-devices`): [roboticslab-uc3m/yarp-devices#205](https://github.com/roboticslab-uc3m/yarp-devices/pull/205)

Merged at https://github.com/roboticslab-uc3m/yarp-devices/commit/304b2455538479d21e2ecfc77678e596f359353d

jgvictores, 2019-03-27:
> * [x]  Accept pull request: [roboticslab-uc3m/teo-developer-manual#50](https://github.com/roboticslab-uc3m/teo-developer-manual/pull/50)

Merged at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/118f4725c86472dc1c8c3782cdde3d49856c6f74

jgvictores, 2019-03-27:
> * [ ]  Accept this pull request: [roboticslab-uc3m/teo-configuration-files#18](https://github.com/roboticslab-uc3m/teo-configuration-files/pull/18)

Merged at https://github.com/roboticslab-uc3m/teo-configuration-files/commit/23703e1e29182a43c491efe04d4f5a9427d6008a

jgvictores, 2019-03-27:
> [x] Update demos (or put issue per repo)
> * https://github.com/roboticslab-uc3m/waiter
> * https://github.com/roboticslab-uc3m/follow-me
> * https://github.com/roboticslab-uc3m/teo-self-presentation

Will be tracked at:
- https://github.com/roboticslab-uc3m/waiter/issues/9
- https://github.com/roboticslab-uc3m/follow-me/issues/46
- https://github.com/roboticslab-uc3m/teo-self-presentation/issues/8

rsantos88, 2019-03-27:
Updated https://github.com/roboticslab-uc3m/teo-main/issues/39#issuecomment-474334416 about **update simulated models**

jgvictores, 2019-03-27:
> [x] Update simulated models (or put issue per repo)

- https://github.com/roboticslab-uc3m/teo-openrave-models/issues/26 (https://github.com/roboticslab-uc3m/teo-openrave-models/pull/25)
- https://github.com/roboticslab-uc3m/teo-simox-models/issues/3
- https://github.com/roboticslab-uc3m/teo_robot/issues/1
- https://github.com/roboticslab-uc3m/teo-gazebo/issues/5

rsantos88, 2019-03-28:
Ok! Can I close this issue or do you want to add more things like the change in kinematics (important issue)?

jgvictores, 2019-03-28:
Yes, please close, and we can actually unblock kinematics! Other issues are all tracked individually. :tada::tada::tada:

closed 2019-03-28 by rsantos88