roboticslab-uc3m/teo-main #41: superbuild repositories for each TEO PC

jgvictores, 2018-03-12:
Extracted from an old version of https://github.com/roboticslab-uc3m/teo-developer-manual :
> We are in fact considering creating specific superbuild repositories for each TEO PC.
* teo-manipulation: For the TEO manipulation PC \(arms and head joint control, JR3 force/torque sensors\).
* teo-locomotion: For the TEO locomotion PC \(legs and torso joint control, Xsens inertial sensor\)
* teo-head: For the TEO head PC \(ASUS XTion Pro Live RGBD sensor, PointGrey RGB camera\).

jgvictores, 2018-03-12:
As with most `idea`s, this is very low-priority. Just made more sense as a GH issue than a comment in the developer manual.

PeterBowman, 2023-03-26:
Keep in mind that manipulation and locomotion are being merged into the same machine: https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/56.