roboticslab-uc3m/teo-main #42: CMake/YCM not finding already installed repos

RaulFdzbis, 2018-03-13:
CMake/YCM not finding already installed repos, the result of doing a cmake over teo_main is the following:

```bash
-- Libraries go to /home/raul/repos/teo-main/build/lib
-- Executables go to /home/raul/repos/teo-main/build/bin
-- YCM found in /usr/local/share/YCM.
-- Package TEO_DEVELOPER_MANUAL not found. Will be downloaded and built.
-- Package TEO_CONFIGURATION_FILES not found. Will be downloaded and built.
-- Package TEO_OPENRAVE_MODELS not found. Will be downloaded and built.
-- Package ROBOTICSLAB_DEVELOPER_MANUAL not found. Will be downloaded and built.
-- Package ROBOTICSLAB_PROJECT_GENERATOR not found. Will be downloaded and built.
-- Package ROBOTICSLAB_INSTALLATION_GUIDES not found. Will be downloaded and built.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/raul/repos/teo-main/build
```

However in my system i have already installed TEO_OPENRAVE_MODELS and TEO_CONFIGURATION_FILES, is this an expected behaviour?

The installation was completed succesfully though.

PeterBowman, 2018-03-13:
> is this an expected behaviour?

Yes, since CMake configuration is not exported for these projects ([ref](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/5e29e7c67e5123d67d528ef277c4868831942d8e/CMakeLists.txt)). Which means that we probably need to review that CMake code at some point...

RaulFdzbis, 2018-03-13:
Okey I understand,

If you agree then I think we can close this issue, anyway I think it could be good to open a new one regarding this in the affected repos, at least just as a way to know that this is something that happens and not forget about it :D

PeterBowman, 2018-03-13:
If you don't mind, I'd rather keep it open and update those repos one by one, which is not that hard and IMHO less tedious than replicating issues everywhere.

RaulFdzbis, 2018-03-13:
Cool!, its ok for me :dancer:

PeterBowman, 2018-03-13:
```
-- Package TEO_DEVELOPER_MANUAL not found. Will be downloaded and built.
-- Package TEO_CONFIGURATION_FILES not found. Will be downloaded and built.
-- Package TEO_OPENRAVE_MODELS not found. Will be downloaded and built.
-- Package ROBOTICSLAB_DEVELOPER_MANUAL not found. Will be downloaded and built.
-- Package ROBOTICSLAB_PROJECT_GENERATOR not found. Will be downloaded and built.
-- Package ROBOTICSLAB_INSTALLATION_GUIDES not found. Will be downloaded and built.
```

Done for all these except ROBOTICSLAB_PROJECT_GENERATOR, which is a bit weird (and perhaps it shouldn't be YCM'ed at all). I'll keep looking for similar cases in ASIBOT and AMOR related repos.

closed 2018-03-13 by PeterBowman