roboticslab-uc3m/teo-main #44: Review and Unify TEO joint names

jgvictores, 2018-03-15:
Review and Unify TEO joint names

jgvictores, 2018-03-15:
For reference, I'd use what is currently called `Simox-names` from:
https://github.com/roboticslab-uc3m/teo-developer-manual/blob/8d2539261ca1755c7040958b9384d99af10dfbe4/assets/motores-motores.csv

jgvictores, 2019-05-23:
Some refs, possibly not too relevant:
- https://discourse.ros.org/t/naming-standards-for-joints-in-humanoid-robots/3992
- http://www.ros.org/reps/rep-0120.html

PeterBowman, 2019-06-26:
Note: "saggital, frontal, axial..." might be misplaced, e.g. current saggital leg joints are actually frontal joints according to https://en.wikipedia.org/wiki/Anatomical_plane.

jgvictores, 2019-10-31:
Related: https://github.com/roboticslab-uc3m/teo-developer-manual/issues/62

PeterBowman, 2019-10-31:
> Note: "saggital, frontal, axial..." might be misplaced, e.g. current saggital leg joints are actually frontal joints according to https://en.wikipedia.org/wiki/Anatomical_plane.

Speaking of planes, yes. Speaking of **axes**, cf. https://www.researchgate.net/publication/270340667 and check the following diagram:

[image]

Moar documents seem to avail this choice of names, see e.g. [this document](https://github.com/roboticslab-uc3m/teo-main/files/3794923/Understanding-Planes-and-Axes-of-Movement.pdf):

> The sagital axis passes horizontally from posterior to anterior and is formed by the intersection of the sagital and transverse planes.
>
> The frontal axis passes horizontally from left to right and is formed by the intersection of the frontal and transverse planes.
>
> The vertical axis passes vertically from inferior to superior and is formed by the intersection of the sagital and frontal planes.

jgvictores, 2019-11-02:
I guess this means we could actually keep something similar to the current names? Namely, regarding root frame:
- X: Sagittal
- Y: Frontal
- Z: Axial

Therefore: `FrontalLeftElbow`, `FrontalTrunk`, right?

jgvictores, 2019-11-02:
PS: I've also thought about simply `LeftElbowY` (`LEY`) and `TrunkY` (`TY`). Partially discarded as these refer to root frame and change depending on previous rotations (but same reasoning may apply to above...).

PeterBowman, 2019-11-02:
I think the sagittal/frontal/axial naming scheme is well established and, since it's actually correct (contrary to what we thought), I'd stick to it rather than switching to XYZ suffixes.

> Therefore: `FrontalLeftElbow`, `FrontalTrunk`, right?

OK.

PeterBowman, 2019-11-04:
Note: review and unify OpenRAVE joint names, too.

jgvictores, 2019-11-04:
> Note: review and unify OpenRAVE joint names, too.

Updated description. May have added some overkill: for now our `.ini`'s do not contain this sort of info.

jgvictores, 2019-11-04:
> * [x]  https://github.com/roboticslab-uc3m/teo-developer-manual

Done at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/e6475723b484c020d46d6251c43d2c61822db6ee

rsantos88, 2019-11-12:
- [x] Openrave models: https://github.com/roboticslab-uc3m/teo-openrave-models

Done at https://github.com/roboticslab-uc3m/teo-openrave-models/pull/35

jgvictores, 2019-11-12:
> Done at [roboticslab-uc3m/teo-openrave-models#35](https://github.com/roboticslab-uc3m/teo-openrave-models/pull/35)

Perfect! I added the PR related to the OpenRAVE model to the [TEO project](https://github.com/orgs/roboticslab-uc3m/projects/6#card-29097675) for visibility, but keeping this issue open for tracking other models. Thanks!!!