roboticslab-uc3m/teo-main #45: Ubuntu 14.04 doesn't support `util-linux 2.23`

rsantos88, 2018-03-19:
Currently we have implemented on `yarp module manager` the command  `/sbin/runuser` to call remotely Teo applications, but with the new downgrade of teo-head (Ubuntu 14.04), when we call this command it can't be found. So, looking for this error, I found that this Ubuntu version doesn't support `util-linux 2.23` (this command appeared in util-linux 2.23). Ubuntu 14.04 ships util-linux 2.20, so it doesn't have this command.

PeterBowman, 2018-03-19:
Not sure if relevant. From https://unix.stackexchange.com/a/169585:

> `runuser` is a recent command, it appeared in [util-linux 2.23](https://mirrors.edge.kernel.org/pub/linux/utils/util-linux/v2.23/v2.23-ReleaseNotes). Ubuntu 14.04 ships [util-linux 2.20](http://packages.ubuntu.com/trusty/util-linux), so it doesn't have this command yet.
>
> `runuser` isn't very useful. Just use su instead. Note that the command `runuser user -c 'cd'` doesn't actually do anything — the scope of cd does not extend to the next call to `runuser`. You'd have to use `runuser user -c 'cd && wget http://file.com'` anyway. Use `su user -c 'cd && wget http://file.com'` instead.

rsantos88, 2018-03-19:
Problem fixed. Replaced command `runuser` by `su` [here](https://github.com/roboticslab-uc3m/teo-configuration-files/pull/10)
Now it's working :)

jgvictores, 2018-03-20:
Fixed by merged https://github.com/roboticslab-uc3m/teo-configuration-files/pull/10

Feel free to close, thanks!

closed 2018-03-20 by rsantos88