roboticslab-uc3m/teo-main #47: Reordenando cables e instalando nuevo router de Teo

jgvictores, 2018-04-06:
_From @rsantos88 on April 6, 2018 12:5_

Estamos @AlvaroMartinezR  y yo reordenando todos los cables y conexiones porque estamos instalando el nuevo router. Teo no funcionará hasta el lunes. Gracias

_Copied from original issue: roboticslab-uc3m/questions-and-answers#58_

rsantos88, 2018-04-09:
**Tareas realizadas:**
- [x] Instalado y configurado el nuevo router central (TEONET) para tener conectividad a internet.
- [x] Limpiado, reordenado y etiquetado todos los cables ethernet que van al router. Ahora, todos los equipos que trabajan con TEO (a excepción del ordenador de @JuanHernandez1993) están conectados a la LAN de TEO.
- [x] Se ha creado un cable nuevo ethernet (categoría 6) y lanzado por debajo del suelo para dar conectividad (a traveś del actual HUB: TP16C) a los 3 ordenadores situados en la parte frontal de TEO (pared del extintor: `teo-nobita`, `teo-suneo`, `teo-benji`) y a `teo-goku` (ordenador actualmente usado por @jgagom ).

**Notas a tener en cuenta:**
El HUB TP16C es bastante antiguo. Tiene una velocidad de 10Mb/s, y el nuevo router dispone de una velocidad de 1000Mb/s, por lo que se está produciendo un cuello de botella bastante importante.

**Tareas pendientes:**
- [ ] Comprar un switch nuevo para sustituir al TP16C que trabaje con mayor velocidad y con un número de puertos considerable (ahora mismo se están usando 4 pero conviene tener más puertos por si se quisiera ampliar el número de equipos conectados a la LAN de TEO)
- [ ] Comprar otro swith nuevo (aunque podría usarse el de @munozyanez pero lo lógico sería que se compre uno) para dar conectividad a `teo-main` y al PC de @JuanHernandez1993 .
- [ ] Configurar los USB-Wifis en TEO para usarlos con el wifi del router cuando sea necesario.
- [ ] Conectar los PCs de TEO con el switch interno de TEO y alimentarlo (para esto, se está esperando a la nueva electrónica)

rsantos88, 2018-04-09:
Actualizo:

**Tareas realizadas:**

- [x] añadida [información](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/appendix/c-teo-network-information.md#teonet-router) de configuración del nuevo router. ([commit1](https://github.com/roboticslab-uc3m/teo-developer-manual/commit/cae7d5eacdde3a5ca3a77c761d8abd681e3f7426), [commit2](https://github.com/roboticslab-uc3m/teo-developer-manual/commit/7c926e0af513d98454dc4d6daabb73da21dd8930)) 

**Añado otra tarea pendiente:**
- [ ] Pedir un HDMI nuevo más largo que el actual para conectar la TV a mi PC (`teo-oliver`)

jgvictores, 2018-04-09:
@rsantos88 @AlvaroMartinezR Gracias.

Por otro lado, me gustaría insistir ahora mismo más que nunca, en lo que se anunció en https://github.com/roboticslab-uc3m/questions-and-answers/issues/47: utilizar `yarp namespace` y un `yarp server` propio cuando no se está trabajando directamente con el robot. Así nos ahorraremos muchos conflictos de nombres/direcciones, que se verían por ejemplo lanzando `teoSim` en varios equipos de la misma red.

jgvictores, 2018-04-10:
En esta línea, acordado con @rsantos88 que moveremos `yarp server` de `teo-main` a `manipulation`. Ventajas:
- teo-main no se quema (está encendido 24/7 y no tiene construcción de servidor)
- teo-main se libera
- menos posibilidad de colisiones de nombres

Consecuencias:
- Rebautizar teo-main

rsantos88, 2018-04-10:
**Actualizo:**

- [x] Conectar los PCs de TEO con el router interno de TEO: `EdgeRouter X (modelo: ER-X)` y alimentarlo (para esto, se está esperando a la nueva electrónica). Configurar dicho router como switch.

Notas para la documentación: 
* Dirección IP actual del dispositivo EdgeRouter X configurado como switch: `2.2.2.10`
* Instrucciones para ser configurado como switch (partiendo de una configuración inicial donde el router tiene la dirección `192.168.1.1` para acceder a su configuración):
```
Full steps would be:
1. Plug into the 1st port, eth0, and log into 192.168.1.1
2. Upgrade the firmware to the latest
3. Add eth1, eth2, eth3, eth4 to switch0 under switch0's config in the GUI.  Do not add eth0.
4 Give an IP address to the switch other than 192.168.1.1, like 192.168.1.10/24 or any ip on your current subnet.
5.When this is done, plug into eth1, and log back into the router under 192.168.1.10
6.Delete the IP address 192.168.1.1 from eth0, and add it to the switch. 

Now all ports are a switch, and accessable from 192.168.1.10

The only limitation of using the ER-X this was is that there is no direct way to use VLANS like a toughswitch, its just a basic dumbsiwtch w/ POE capability now.
```
**ToDo:**

- [ ] El cable ethernet que va del nuevo switch a `teo-head` no funciona. Hay que rehacer el cable.

rsantos88, 2018-04-11:
El cable ethernet de `teo-head` ya funciona correctamente. Gracias @AlvaroMartinezR

jgvictores, 2018-04-12:
> En esta línea, acordado con @rsantos88 que moveremos yarp server de teo-main a manipulation. 

Cambiaré el namespace del yarpserver de Teo a /teo, ok?
Tranqui tods que actualizaré la doc. :-)

AlvaroMartinezR, 2018-04-13:
Desde que hemos cambiado el servidor de sitio y el router, a la hora de lanzar una trayectoria basada en posiciones TEO ejecuta a saltos, incluso el codigo se queda detenido. @munozyanez y yo hemos observado que cuando el codigo se detiene, la tarjeta de CAN Bus se pone en rojo. Cuando vuelve a ejecutar ordenes, vuelve a verde.

rsantos88, 2018-04-17:
> Desde que hemos cambiado el servidor de sitio y el router, a la hora de lanzar una trayectoria basada en posiciones TEO ejecuta a saltos, incluso el codigo se queda detenido. @munozyanez y yo hemos [...]

**Observación:** la velocidad de conexión del nuevo router es 1000Mb/s al igual que el del swith interno de Teo. Al realizar un ping a manipulation, el valor más alto obtenido es 0.173ms 

**Idea:** Para descartar que se tratase de un problema de velocidad de transmisión con el router o cuello de botella, lanzar la trayectoria ejecutándose directamente en el PC de `teo-manipulation` o `teo-locomotion` y comprobar los resultados.

munozyanez, 2018-04-17:
Gracias @rsantos88 , mañana probaremos lo que comentas.

rsantos88, 2018-04-23:
> - [ ] Comprar un switch nuevo para sustituir al TP16C que trabaje con mayor velocidad y con un número de puertos considerable (ahora mismo se están usando 4 pero conviene tener más puertos por si se quisiera ampliar el número de equipos conectados a la LAN de TEO)

Se ha pensado en utilizar el el antiguo router de Teo **TP-LINK** sustituir el **TP16C**

rsantos88, 2018-04-23:
añadida info del nuevo router interno de Teo: https://github.com/roboticslab-uc3m/teo-developer-manual/commit/a1d132cab7b368a75769c706186f342d2c7e8c30

rsantos88, 2018-04-23:
Router TP-LINK intalado y configurado.
Nueva información del mismo añadida [aquí](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/appendix/c-teo-network-information.md#tl-wr2543nd-wan-router-network-to-interconnect-computers-with-teonet)

rsantos88, 2018-04-23:
TP-LINK configurado como [switch](https://github.com/roboticslab-uc3m/teo-developer-manual/blob/master/appendix/c-teo-network-information.md#tl-wr2543nd-router-configured-as-switch)
Recordar que para configurarlo como switch hay que deshabilitar DHCP y asignar una IP LAN estática que esté dentro de nuestra red (actualmente la 2.2.2.2)
Nota: el puerto WAN no permite utilizarse al estar en modo switch, por lo que solo tiene posibilidad de conectar 3 PCs.

rsantos88, 2018-11-06:
También se ha configurado la Wifi del router central de Teo para poder acceder a la red de esta forma.
 TEONET-2G (up to 800Mbps)
 TEONET-5G (up to 1733Mbps)
 TEONET-60G (up to 4.6Gbps)

Cierro issue

closed 2018-11-06 by rsantos88