roboticslab-uc3m/teo-main #50: Docker for TEO Development

ajardon, 2021-09-20:
Due ALMA EU project needs, the idea is to develop a docker to make easier to develop and deploy algorithms that has to be tested on TEO platform (at least at simulation level...)

PeterBowman, 2021-09-20:
Kind of related to https://github.com/roboticslab-uc3m/vision/issues/107#issuecomment-835229150 on a more general basis.

PeterBowman, 2023-03-26:
What kind of functionality should this Docker image provide? The request is way too broad and unspecific, besides this might not be the right place for it (can't tell from the description). Recently, we have published a container elsewhere targetting the ALMA project, so I believe this issue might be superseded:

* https://github.com/roboticslab-uc3m/gymnasium-playground
* https://github.com/roboticslab-uc3m/gymnasium-playground/pkgs/container/gymnasium-playground

Closing as invalid, for now. Please provide any details if reopened.

closed 2023-03-26 by PeterBowman

jgvictores, 2023-11-19:
From https://asrob.uc3m.es/tutoriales/software/docker.html#algunos-images-buenos

- <https://github.com/roboticslab-uc3m/teo-demos-misc/pkgs/container/teo-demos-misc> (`ghcr.io/roboticslab-uc3m/teo-demos-misc`) -> TEO OpenRAVE+YARP+Kinematics+...