roboticslab-uc3m/teo-main #51: Gait demo using iCub's walking controllers

PeterBowman, 2021-09-28:
On 28th September, an attempt was made to launch a gait trajectory Luís had prepared beforehand using https://github.com/robotology/walking-controllers and reworked TEO's SDF/URDF models. The dataset (involving both legs, both arms and the trunk) and the config file are uploaded here: [teo-gait-20210927.zip](https://github.com/roboticslab-uc3m/teo-main/files/7247547/teo-gait-20210927.zip). We used [examplePlaybackThread](https://github.com/roboticslab-uc3m/tools/blob/1ef26ed09258a238fd8d57a678c9518912048e6c/examples/cpp/examplePlaybackThread/examplePlaybackThread.cpp) from the tools repo in order to issue the joint commands.

The plan for said session was as follows:
* A dry-run of the offline-saved trajectory on the real robot (i.e. while lifted by its crane).
* Same trajectory, robot actually walking on the floor.
* Online-generated trajectory using FT sensor feedback.

We only managed to fulfill the first point, at least most of it. A control error was triggered in ID4 (right frontal hip). Currently, we are facing other blocking issues, though:
* CUI111 (frontal left ankle) was reporting weird values on start, usually around 60-67 degrees. This is probably a known issue, I think we already experienced a similar behavior in the past. On this occasion, Luís noticed the encoder was visibly damaged and it stopped responding altogether (i.e. no response from polling) later during the experiments. I suggest to replace this CUI.
* Prior to that, maybe related or not, faulty behavior was observed on left leg bus: https://github.com/roboticslab-uc3m/yarp-devices/issues/246#issuecomment-926791504. However, this time no bus restarts seemed to occur, but we did observe an iPOS error pointing at a damaged connection with the relative encoder. If CUI111 is not the underlying cause for this, the wiring and connectors along the left leg bus should be closely inspected.

cc @smcdiaz

traversaro, 2021-10-01:
Just FYI, another repo you may want to check out (even if I don't know if it is suitable for what you want to do) is https://github.com/ami-iit/bipedal-locomotion-framework .

PeterBowman, 2021-10-01:
Thank you, @traversaro! Regarding walking-controllers, we've been inspecting the generated trajectory more closely and found out that certain points result in excessive joint velocities despite having configured much lower velocity limits on the robot. I can see the only places where those limits play a role seem to be the IK controllers: [qpOASES](https://github.com/robotology/walking-controllers/blob/dcc2698e3d16adb5c78c973a27a359f0986ae535/src/WholeBodyControllers/src/QPInverseKinematics_qpOASES.cpp#L45-L60) and [osqp](https://github.com/robotology/walking-controllers/blob/dcc2698e3d16adb5c78c973a27a359f0986ae535/src/WholeBodyControllers/src/QPInverseKinematics_osqp.cpp#L52-L74).

traversaro, 2021-10-02:
Hi @PeterBowman ! Feel free to open an issue in the specific repo ( https://github.com/robotology/walking-controllers ) with the details of how you use the software and the specific version you use so that it may be easier for mantainers to comment on this, thanks.

PeterBowman, 2022-04-28:
https://github.com/Luis93A/walking-controllers

PeterBowman, 2022-05-27:
Nice guides found on Luís' PC:
- [walking-controller-guide.pdf](https://github.com/roboticslab-uc3m/teo-main/files/8785628/walking-controller-guide.pdf)
- [walking_controller_teo_readme.txt](https://github.com/roboticslab-uc3m/teo-main/files/8785629/walking_controller_teo_readme.txt)

PeterBowman, 2022-09-12:
I'm uploading here the nice gait log visualizer for TEO coded by @ThomasAiraud: [TeoLogVisualizer.zip](https://github.com/roboticslab-uc3m/teo-main/files/9547206/TeoLogVisualizer.zip). I have stripped the /linux-amd64/java/ directory from this ZIP since it's a full JVM and thus way too heavy to be included. The Java version used therein was Temurin-17.0.2+8. I believe that manually installing the appropriate JVM via apt and tweaking the JAVA_HOME and related env vars will be enough to make this work. There is also a full ZIP uploaded to gdrive [here](https://drive.google.com/file/d/1LQb8PCnkG5G7nipyijks5YmfpVMbApb0/view).

<details>
<summary>Guide to Use the TEO Log Visualizer (README.md)</summary>

# Guide to Use the TEO Log Visualizer

This application is created to visualize the data generated by the WalkingLoggerModule of TEO.

# Overview

- [Directory Structure](#directory-structure)
- [Installation](#installation)
- [How to Run TEO Log Visualizer](#how-to-run-teo-log-visualizer)
- [Graphic User Interface](#graphic-user-interface)
- [Uses of TEO Log Visualizer](#uses-of-teo-log-visualizer)

# Directory Structure

```markdown
📦TeoLogVisualizer
 ┣ 📂linux-amd64
 ┃ ┣ 📂lib
 ┃ ┣ 📂source
 ┃ ┣ 📂java
 ┃ ┗ 📜TeoLogVisualizer
 ┣ 📜README.md
 ┣ 📜Makefile
 ┣ 📜Button.pde
 ┣ 📜Data.pde
 ┣ 📜TeoLogVisualizer.pde
 ┗ 📜Visualizer.pde
```

📦TeoLogVisualizer is the main project directory containing the Processing files.

📂linux-amd64 is the directory generated by the Processing project for Linux (Intel 64-bit).\
It contains the bash file 📜TeoLogVisualizer which run the application.

# Installation

## Install

- Download and extract the archive. 
- Run `Makefile`

``` sh
        make install
```

## Remove

- Run `Makefile`

``` sh
        make remove
```

# How to Run TEO Log Visualizer

1. On a terminal run `TeoLogVisualizer`:

    ```sh
    TeoLogVisualizer
    ```

2. Choose the file with the data you want to visualize from the file explorer.

# Graphic User Interface

[image]

## Buttons

□ \*Clear params\*\
Clear all the parameters selected and reset the zoom.

□ Add another file\
Allow you to select another data file with the file explorer.

□ Match start of movement\
Allow you to select another data file with the file explorer.

□ dcm_x\
Example of one parameter from the data file, you can select it to display it on the visualizer.

## Keyboard key binding

▼ Zoom in\
▲ Zoom out\
◀ Move to the left\
▶ Move to the right

## Information displayed

- The evolution of the selected parameters.
- The ordinate axis is values of the signal.
- The abscissa axis is the time of the signal.

# Uses of TEO Log Visualizer

What you can do with TEO Log Visualizer is:

- Display the evolution of parameters from the simulation.
- Display the evolution of the same paramenters of two different simulation.
- Put the two different signals of the same parameter in phase and compare them.

## Warning

You only can match signals from simulation in which TEO walked.
</details>

Merci, Thomas !